CadQuery/cadquery #508: Bug: Offset2D does not work for a single line

blooop, 2020-11-16:
I would like to create a capsule shape by offsetting from a line by a fixed radius, but I get an error "Null TopDS_Shape object"

I have found a workaround where I create an arc with a very large radius to approximate a straight line and that works.   Am I doing something wrong or is this a bug?

```
import cadquery as cq

lineLen = 1
arcHack = False
r = cq.Workplane("XY")
if arcHack:
    r = r.sagittaArc((0, lineLen), 0.01)
else:
    r = r.line(0,lineLen)
r= r.offset2D(0.1)
r =r.extrude(0.2)
```

adam-urbanczyk, 2020-11-16:
That sounds like a bug, here is a nicer workaround:
```python
import cadquery as cq

lineLen = 1
r = cq.Workplane("XY")
r = r.line(0,lineLen).close()
r= r.offset2D(.1)
```
you can also use the `slot2D` method BTW.

blooop, 2020-11-16:
Hi

Thanks for the workaround.  Just for my understanding, is this essentially creating another line from (0,lineLen) back to the origin so there are 2 lines on top of eachother? 

I was hoping to use the line and save the start and end vertices so I could define hole cutting positions after.  Is that possible with slot2D?  (from my testing just now I was not able to get it to work, but I also can't get it to work with line).  Is this expected behaviour or do you think it is related to the bug? (It returns the same error message)

```ValueError: Null TopoDS_Shape object```

my failed attempt a cutting holes defined by the first line:
```
import cadquery as cq

lineLen = 3.14
r = cq.Workplane("XY")
r = r.line(0, lineLen).close().tag("holes")  # want to save vertex (0,0) and (0,lineLen) to define holes later on
r = r.offset2D(0.2)
r = r.extrude(0.2)
for o in r.vertices(tag="holes").objects:
    print(o.Center())  # I seem to have tagged the two vertices correctly
r = r.vertices(tag="holes").hole(0.1, -1)  # does not cut as expected
```

marcus7070, 2020-11-16:
You were almost there. 

```python
import cadquery as cq

lineLen = 3.14
r = cq.Workplane("XY")
r = r.line(0, lineLen).close().tag("holes")  # note this line is in the XY plane, ie. with z=0
r = r.offset2D(0.2)
r = r.extrude(0.2)
r = r.faces(">Z").workplane() # This is the top plane, with z=0.2
r = r.vertices(tag="holes").hole(0.1, 1)  # select vertices with z=0 and cut in the negative z direction
```

To get this example to work either:

- invert the first workplane, `r = cq.Workplane("XY").workplane(invert=True)`, or
- extrude in the negative direction, `r = r.extrude(-0.2)`.

Now the selected vertices are in the workplane and the hole operation will cut in the positive z direction.

adam-urbanczyk, 2020-11-17:
> Thanks for the workaround. Just for my understanding, is this essentially creating another line from (0,lineLen) back to the origin so there are 2 lines on top of eachother?

Your understanding is correct. They are also topologically connected, i.e. they form a closed wire.