CadQuery/cadquery #511: Assemblies Best Practices

blooop, 2020-11-17:
Hi,

I've run the example code on how to set up an assembly:
https://cadquery.readthedocs.io/en/latest/primer.html?#assemblies

but I can't really find any examples on how to use constraints.  

This is my attempt:
```
import cadquery as cq

def createLink(length=5, width=2, thickness=1, holeD=0.5):
    r = cq.Workplane("XY")
    r = r.line(length, 0).close().tag("holes")
    r = r.offset2D(width)
    r = r.extrude(-thickness)
    r = r.faces(">Z").workplane()
    r = r.vertices(tag="holes").hole(holeD)
    # not sure how to do this bit, but want to tag the edge in the center of each hole
    # and tag the other hole hole2
    # r = r.edges("<X").tag("hole1Axis")
    # r = r.edges(">X").tag("hole2Axis")
    return r

part1 = createLink(5)
part2 = createLink(7, 3)

assy = cq.Assembly()
assy = assy.add(part1)
assy = assy.add(part2, color=cq.Color(0, 0, 1, 0.5))
assy = assy.constrain((part1, part2), (part1.faces(">Z"), part2.faces("<Z")), "Plane")  # does not work, ideally the two faces would touch
assy = assy.constrain((part1, part2), (part1.edges(tag="hole1Axis"), part2.edges(tag="hole2Axis")), "Axis")
# does not work, ideally the two hole would be coaxial

show_object(assy)
```

I have found:
https://github.com/cqparts/cqparts

Is there more documentation I have not found on how to set up assemblies with constraints?  Is using the assembly module within cq the preferred method or is cqparts more comprehensive?

Thanks

adam-urbanczyk, 2020-11-17:
The assy system in CQ is really new so there are no docs so far. I never used cqparts so I cannot comment.

This is already covered in #469 so I propose to continue any discussion on this topic there.

 BTW: Here https://github.com/marcus7070/spindle-assy-example/tree/assembly you can find an example created by @marcus7070 .

closed 2020-11-17 by adam-urbanczyk

jmwright, 2020-11-18:
@blooop Please continue the discussion over on #469 but it should be noted for anyone who finds this later that cqparts is currently not maintained.

blooop, 2020-11-18:
Thanks for the tips.  I will check out that repo and thread to get some inspiration.