CadQuery/cadquery #512: broken by OCP 7.5-RC1 because of missing font?

greyltc, 2020-11-19:
This seems not to work with https://github.com/CadQuery/OCP/releases/tag/7.5-RC1
Something about a missing font:

```
$ pytest -v --cov
============================================================================================================ test session starts =============================================================================================================
platform linux -- Python 3.8.6, pytest-6.1.2, py-1.9.0, pluggy-0.13.1 -- /usr/bin/python
cachedir: .pytest_cache
PyQt5 5.15.1 -- Qt runtime 5.15.1 -- Qt compiled 5.15.1
rootdir: /home/REDACTED/git/python-cadquery-git/src/cadquery
plugins: qt-3.3.0, mock-2.0.0, cov-2.10.1, dash-1.16.1
collected 0 items / 9 errors                                                                                                                                                                                                                 

=================================================================================================================== ERRORS ===================================================================================================================
__________________________________________________________________________________________________ ERROR collecting tests/test_assembly.py ___________________________________________________________________________________________________
ImportError while importing test module '/home/REDACTED/git/python-cadquery-git/src/cadquery/tests/test_assembly.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
/usr/lib/python3.8/importlib/__init__.py:127: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
tests/__init__.py:1: in <module>
    from cadquery import *
cadquery/__init__.py:3: in <module>
    from .occ_impl.shapes import (
cadquery/occ_impl/shapes.py:149: in <module>
    from OCP.Font import (
E   ImportError: cannot import name 'Font_BRepTextBuilder' from 'OCP.Font' (unknown location)
_________________________________________________________________________________________________ ERROR collecting tests/test_cad_objects.py _________________________________________________________________________________________________
ImportError while importing test module '/home/REDACTED/git/python-cadquery-git/src/cadquery/tests/test_cad_objects.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
/usr/lib/python3.8/importlib/__init__.py:127: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
tests/__init__.py:1: in <module>
    from cadquery import *
cadquery/__init__.py:3: in <module>
    from .occ_impl.shapes import (
cadquery/occ_impl/shapes.py:149: in <module>
    from OCP.Font import (
E   ImportError: cannot import name 'Font_BRepTextBuilder' from 'OCP.Font' (unknown location)
__________________________________________________________________________________________________ ERROR collecting tests/test_cadquery.py ___________________________________________________________________________________________________
ImportError while importing test module '/home/REDACTED/git/python-cadquery-git/src/cadquery/tests/test_cadquery.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
/usr/lib/python3.8/importlib/__init__.py:127: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
tests/__init__.py:1: in <module>
    from cadquery import *
cadquery/__init__.py:3: in <module>
    from .occ_impl.shapes import (
cadquery/occ_impl/shapes.py:149: in <module>
    from OCP.Font import (
E   ImportError: cannot import name 'Font_BRepTextBuilder' from 'OCP.Font' (unknown location)
____________________________________________________________________________________________________ ERROR collecting tests/test_cqgi.py _____________________________________________________________________________________________________
ImportError while importing test module '/home/REDACTED/git/python-cadquery-git/src/cadquery/tests/test_cqgi.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
/usr/lib/python3.8/importlib/__init__.py:127: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
tests/__init__.py:1: in <module>
    from cadquery import *
cadquery/__init__.py:3: in <module>
    from .occ_impl.shapes import (
cadquery/occ_impl/shapes.py:149: in <module>
    from OCP.Font import (
E   ImportError: cannot import name 'Font_BRepTextBuilder' from 'OCP.Font' (unknown location)
__________________________________________________________________________________________________ ERROR collecting tests/test_exporters.py __________________________________________________________________________________________________
ImportError while importing test module '/home/REDACTED/git/python-cadquery-git/src/cadquery/tests/test_exporters.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
[... 63 lines cut ...]
Name                            Stmts   Miss Branch BrPart  Cover
-----------------------------------------------------------------
cadquery/__init__.py               11      9      0      0    18%
cadquery/occ_impl/__init__.py       0      0      0      0   100%
cadquery/occ_impl/geom.py         453    273     94      1    33%
cadquery/occ_impl/shapes.py       903    865    234      0     3%
tests/__init__.py                  32     31      6      0     3%
-----------------------------------------------------------------
TOTAL                            1399   1178    334      1    13%

========================================================================================================== short test summary info ===========================================================================================================
ERROR tests/test_assembly.py
ERROR tests/test_cad_objects.py
ERROR tests/test_cadquery.py
ERROR tests/test_cqgi.py
ERROR tests/test_exporters.py
ERROR tests/test_importers.py
ERROR tests/test_jupyter.py
ERROR tests/test_selectors.py
ERROR tests/test_workplanes.py
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 9 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
============================================================================================================= 9 errors in 1.01s ==============================================================================================================

```
Any ideas for a fix?

greyltc, 2020-11-19:
I dropped the `text()` function from the codebase entirely and my cadquery is working again: https://github.com/greyltc/cadquery/commit/48bab8ad9e176590f67980a263262b4c6b0b3198

adam-urbanczyk, 2020-11-19:
CQ works with 7.4 for now (this is specified in the conda recipe AFAIR).

marcus7070, 2021-04-06:
Pretty sure this is solved by OCP 7.5, please re-open @greyltc if that's not the case.

closed 2021-04-06 by marcus7070

greyltc, 2021-04-06:
Yep. This is fixed.