CadQuery/cadquery #516: OCP.Standard.Standard_OutOfRange: NCollection_Sequence::ChangeValue

ahupp, 2020-11-20:
cadquery is version 2.1.dev0, python 3.8.6

I'm trying to turn an image into a heightmap on a surface by extruding the individual "pixels".   My expectation was that extruding partially overlapping circles from a surface will result in the union of those shapes, but that doesn't seem to be the case.

This test script:

`
import cadquery as cq
hmap = [(1.1525965009719523, -0.3269925020827549), (1.0242988058872533, -0.28200499861149675)]

r = cq.Workplane("XY" ).box(4, 4, 0.5).faces(">Z")
r = r.pushPoints(hmap)
r = r.circle(.1)
r = r.extrude(1.0)
`

produces this error:

`
Traceback (most recent call last):
  File "cqheightmap.py", line 33, in <module>
    r = r.extrude(1.0)
  File "/home/ahupp/.conda/envs/cq-editor/lib/python3.8/site-packages/cadquery/cq.py", line 2703, in extrude
    newS = newS.clean()
  File "/home/ahupp/.conda/envs/cq-editor/lib/python3.8/site-packages/cadquery/cq.py", line 3539, in clean
    cleanObjects = [
  File "/home/ahupp/.conda/envs/cq-editor/lib/python3.8/site-packages/cadquery/cq.py", line 3540, in <listcomp>
    obj.clean() if isinstance(obj, Shape) else obj for obj in self.objects
  File "/home/ahupp/.conda/envs/cq-editor/lib/python3.8/site-packages/cadquery/occ_impl/shapes.py", line 343, in clean
    upgrader.Build()
OCP.Standard.Standard_OutOfRange: NCollection_Sequence::ChangeValue
`

I suspect this has something to do with the extruded cylinders partially overlapping, but when I try to construct examples like that I get a series of different, surprising results.

With `[(0.0, 0.0), (.05, .05)]` I get some kind of difference operation:

[image]

And with `[(0.0, 0.0), (.01, .01)]` I get these overlapping hollow cylinders, rather than solid cylinders:

[image]

My intuition for what extrude() does is clearly wrong, but I'm not sure how to figure out the right behavior here.

adam-urbanczyk, 2020-11-20:
Extruded profiles cannot intersect, two concentric circles with different radii will give you a thick pipe.

To construct your solid you might want to use `each` with a custom callback (see extending cq in the docs).

jmwright, 2020-11-20:
Here's a link to that section in the docs: https://cadquery.readthedocs.io/en/latest/extending.html

ahupp, 2020-11-20:
> Extruded profiles cannot intersect

Thanks!  If it's not allowed, is it possible to explicitly raise an error when this happens?  I didn't see a mention of that limitation in the extrude docs:

https://pythonhosted.org/cadquery/classreference.html#cadquery.Workplane.extrude

adam-urbanczyk, 2020-11-21:
PRs regarding documentation are welcome, I'm not sure one ones to do a self-interesection check on every operation. BTW: here is a up to date doc link: https://cadquery.readthedocs.io/en/latest/?badge=latest

Here is a complete example if you still need it:
```python
import cadquery as cq
hmap = [(1.1525965009719523, -0.3269925020827549), (1.0242988058872533, -0.28200499861149675)]

base = cq.Workplane().box(4, 4, 0.5)
r =  (base
      .faces(">Z").workplane()
      .pushPoints(hmap)
      .eachpoint(
          lambda p: cq.Workplane().circle(.1).extrude(1).val().locate(p))
      .union(base)
     )
```

tanius, 2021-01-31:
I was missing a way to extrude overlapping wires (i.e. 2D unions) as well. Since [FreeCad supports unions of 2D faces](https://forum.freecadweb.org/viewtopic.php?t=13863) this is supported in the OpenCascade CAD kernel. So at first I tried union'ing faces in CadQuery, converting the result to a wire and extruding that. The face and wire would look fine but I could not get it to work due to a "Null Topo_DS object" error. Example:

```
result = (
    cq.Workplane("XY")
    .add( cq.Face.makeFromWires(cq.Workplane("XY").rect(40, 40).val()) )
    .add( cq.Face.makeFromWires(cq.Workplane("XY").circle(25).val()) )
    .combine(clean = True)
    .faces().wires().toPending().extrude(10)
)
```
Is 2D union'ing possible with such an approach?

At this point I had no better idea than to extrude the wires, union the solids, and convert the bottom face of the resulting solid back into a wire. That works, but is obviously a hacky workaround, and will be slow in case you have many objects. I wasn't sure if I should even post this here, but … there it goes, maybe it's useful to somebody. It's a [CadQuery plugin](https://cadquery.readthedocs.io/en/latest/extending.html#extending-cadquery-plugins) that you can call like any other Workplane class method after registering it.

```
def union_pending_wires(self):
    """
    CadQuery plugin that replaces all pending wires with a wire that is their 2D union. It requires all 
    pending wires to be co-planar. This supplements CadQuery's Workplane::combine() and 
    Workplane::consolidateWires() and Wire::combine(), which cannot deal with intersecting wires yet.
    :return: A Workplane object with the combined wire on the stack (besides nothing else) and in 
        its pending wires (besides nothing else).
    .. todo:: Enforce that all wires must be co-planar, raising an error otherwise.
    """

    wires = [obj for obj in self.ctx.pendingWires if isinstance(obj, cq.Wire)]
    if len(wires) < 2: return self # Nothing to union for 0 or 1 pending wires.

    extrude_direction = wires[0].normal()
    solids = (
        cq.Workplane("XY")
        # Create a workplane coplanar with the wires, as this will define the extrude() direction.
        .add(cq.Face.makeFromWires(wires[0]))
        .workplane()
    )

    for wire in wires:
        solids = solids.add(wire).toPending().extrude(1)

    union_wire = (
        solids
        .combine() # 3D union of all the solids.
        # Select the bottom face, as that contains the wires in their original local z position.
        .faces(cq.DirectionMinMaxSelector(extrude_direction, directionMax = False))
        .wires()
    )

    # We have to build the result in the actual object that will be returned to have control over 
    # its pending wires and edges. A "return self.newObject(combined_wire.objects)" would simply 
    # copy over the pending wires and edges of the given object.
    result = self.newObject(union_wire.objects)
    result.ctx.pendingWires = [union_wire.val()] # Replace existing pending wires with union wire.

    return self.newObject(result.objects)

# Usage example
cq.Workplane.union_pending_wires = union_pending_wires
result = (
    cq.Workplane("XY")
    .rect(40,40)
    .circle(24)
    .center(-20, 0).rect(20, 16)
    .union_pending_wires()
    .extrude(15)
)
show_object(result)
```