CadQuery/cadquery #518: Possibilities for converting TopoDS geometry from OCP to OCC

Krande, 2020-11-22:
Hi,

I have some designs that I have written in cadquery that I used to be able to export to an IFC file using [Ifcopenshell](https://github.com/IfcOpenShell/IfcOpenShell). After updating cadquery to the latest version (my previous version was pretty old :) ) I noticed that ifcopenshell did not accept the OCP.TopoDS.TopoDS_Compound objects (which I assume is new after the shift from the OCC cad kernel to OCP?). 

My initial thought was to export STEP files using the OCP kernel and re-import into the OCC kernel, but it would be nice to have some direct conversion of geometry in-memory. 

I saw in the OCC branch's `BRepTools_ShapeSet` has a `WriteToString()` method that I thought I could find in the same OCP equivalent, but unfortunately I was unable to find it. Is it wrapped or are there any other ways of doing this?

So here is what I thought I could use in OCP (but unfortunately the `WriteToString()` method was not found). 

```
# shape = <TopoDS_Compound>

from OCP.BRepTools import BRepTools_ShapeSet
from OCP.TopoDS import TopoDS_Compound

shapes = BRepTools.BRepTools_ShapeSet()
shapes.Add(shape)
occ_string = shapes.WriteToString()
```

If anyone have any suggestions for how this could be done I would really appreciate it? Also let me know if there is anything I can do to advance this issue!

Best Regards
Kristoffer

adam-urbanczyk, 2020-11-24:
`BRepTools_ShapeSet` has a `Write` method as far as I can tell. The one you are referring to does not exist in the OCCT docs (https://old.opencascade.com/doc/occt-7.4.0/refman/html/class_b_rep_tools___shape_set.html).

At the moment this method does write to `std::ostream` which is not wrapped by pybind11. There are more methods using `std::ostream` (`TopoDS_Shape.DumpJson`) so having this implemented would be interesting. I managed to find an example https://gist.github.com/asford/544323a5da7dddad2c9174490eb5ed06 which I'd be willing to integrate into OCP. If you are up to the task please open a PR in the OCP repo, otherwise you'll have to wait for me or someone else from the community implementing it.

Krande, 2020-11-24:
If what you are referring to requires doing anything with c++ and pybind11 it will take me a bit of time to get the local development setup started. So maybe others might get this job done a lot faster. But I'll give it a shot anyway and create a PR as soon as I have something that works!

Best Regards
Kristoffer

adam-urbanczyk, 2020-11-24:
Yes, pybind11/C++ will be required.

adam-urbanczyk, 2023-03-17:
It is supported now, see https://github.com/CadQuery/cadquery/blob/2d355177871c0b1a275640659ba203cf3cdf267c/cadquery/occ_impl/shapes.py#L508

closed 2023-03-17 by adam-urbanczyk