CadQuery/cadquery #523: .loft() chooses "incorrect" wire?

yarikoptic, 2020-11-27:
I am trying to achieve something like what is described in https://stackoverflow.com/questions/61186339/how-to-iteratively-loft-wires-in-cadquery :+1: 

```python
import cadquery as cq
wp = cq.Workplane("XY").rect(1,2).workplane()
result = None
for i in range(0,5):
    wp2 = wp.transformed(offset=cq.Vector(0, -0.5, 1.0),rotate=cq.Vector(10, 0, 0)).rect(1,2).workplane()
    if result == None:
        result = wp2.loft(combine=True)
    else:
        nextpart = wp2.loft(combine=True)
        result = result.union(nextpart)
    wp = wp.transformed(offset=cq.Vector(0, -0.5, 1.0),rotate=cq.Vector(10, 0, 0)).rect(1,2).workplane()
show_object(result, options=dict(alpha=0.5,color='red'))
```

(in my case it would be not just a `.rect` but a `.polyline` with different at each step shape), but for my this example renders differently than on that page, and seems incorrect -- wide side face connects to the wire of 2nd rect instead of the one before: 

[image]

re-installed clean using `conda create -n cq-editor2 -c cadquery -c conda-forge cq-editor=master` to the same result.

<details>
<summary>versions I have installed in conda</summary> 

```shell
$> conda list    
# packages in environment at /home/yoh/anaconda-5.2.0-2.7/envs/cq-editor2:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
alabaster                 0.7.12                     py_0    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
argh                      0.26.2          pyh9f0ad1d_1002    conda-forge
astroid                   2.4.2            py38h32f6830_1    conda-forge
async_generator           1.10                       py_0    conda-forge
atomicwrites              1.4.0              pyh9f0ad1d_0    conda-forge
attrs                     20.3.0             pyhd3deb0d_0    conda-forge
autopep8                  1.5.4              pyh9f0ad1d_0    conda-forge
babel                     2.9.0              pyhd3deb0d_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.1                      py_0    conda-forge
black                     20.8b1                     py_1    conda-forge
bleach                    3.2.1              pyh9f0ad1d_0    conda-forge
brotlipy                  0.7.0           py38h8df0ef7_1001    conda-forge
bzip2                     1.0.8                h516909a_3    conda-forge
c-ares                    1.17.1               h36c2ea0_0    conda-forge
ca-certificates           2020.11.8            ha878542_0    conda-forge
cadquery                  master                    py3.8    cadquery
certifi                   2020.11.8        py38h578d9bd_0    conda-forge
cffi                      1.14.4           py38ha312104_0    conda-forge
chardet                   3.0.4           py38h924ce5b_1008    conda-forge
click                     7.1.2              pyh9f0ad1d_0    conda-forge
cloudpickle               1.6.0                      py_0    conda-forge
cq-editor                 master                    py3.8    cadquery
cryptography              3.2.1            py38h7699a38_0    conda-forge
curl                      7.71.1               he644dc0_8    conda-forge
dataclasses               0.7                pyhb2cacf7_7    conda-forge
dbus                      1.13.6               hfdff14a_1    conda-forge
decorator                 4.4.2                      py_0    conda-forge
defusedxml                0.6.0                      py_0    conda-forge
diff-match-patch          20200713           pyh9f0ad1d_0    conda-forge
docutils                  0.16             py38h924ce5b_2    conda-forge
double-conversion         3.1.5                he1b5a44_2    conda-forge
eigen                     3.3.8                h0efe328_0    conda-forge
entrypoints               0.3             pyhd8ed1ab_1003    conda-forge
expat                     2.2.9                he1b5a44_2    conda-forge
[... 190 lines cut ...]
wrapt                     1.11.2           py38h1e0a361_1    conda-forge
wurlitzer                 2.0.0            py38h32f6830_2    conda-forge
x264                      1!152.20180806       h14c3975_0    conda-forge
xorg-kbproto              1.0.7             h14c3975_1002    conda-forge
xorg-libice               1.0.10               h516909a_0    conda-forge
xorg-libsm                1.2.3             h84519dc_1000    conda-forge
xorg-libx11               1.6.12               h516909a_0    conda-forge
xorg-libxau               1.0.9                h14c3975_0    conda-forge
xorg-libxdmcp             1.1.3                h516909a_0    conda-forge
xorg-libxext              1.3.4                h516909a_0    conda-forge
xorg-libxt                1.2.0                h516909a_0    conda-forge
xorg-xextproto            7.3.0             h14c3975_1002    conda-forge
xorg-xproto               7.0.31            h14c3975_1007    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
yaml                      0.2.5                h516909a_0    conda-forge
yapf                      0.30.0             pyh9f0ad1d_0    conda-forge
zeromq                    4.3.3                h58526e2_3    conda-forge
zipp                      3.4.0                      py_0    conda-forge
zlib                      1.2.11            h516909a_1010    conda-forge
zstd                      1.4.5                h6597ccf_2    conda-forge

```
</details>

PS if someone could share a code snippet for similar, no rotation just vertically `loft-ing` up polylines -- would be VERY much appreciated, I am cadquery newbee and still fighting how to do it properly

Andy-Valentine, 2020-11-27:
Hi,

it seems like you are facing the same issue as I described in #520. Loft sometimes flips wires which you assign to it resulting in self-intersecting lofts. And the use of .union() on self-intersecting pieces produces glitches like you described.

I made a workaround which you can lookup in the other thread. Just need to replace **.rect()** with **.polyline()**

adam-urbanczyk, 2020-11-27:
This is not related to #520, you can make your code work like so (`clean=False` in the union):

```python
import cadquery as cq
wp = cq.Workplane("XY").rect(1,2).workplane()
result = None

for i in range(0,5):
    wp2 = wp.transformed(offset=cq.Vector(0, -0.5, 1.0),rotate=cq.Vector(10, 0, 0)).rect(1,2).workplane()
    if result == None:
        result = wp2.loft(combine=True)
    else:
        nextpart = wp2.loft(combine=True)
        result = result.union(nextpart,clean=False)
    wp = wp.transformed(offset=cq.Vector(0, -0.5, 1.0),rotate=cq.Vector(10, 0, 0)).rect(1,2).workplane()
```

adam-urbanczyk, 2020-11-27:
@yarikoptic note that you can also use sweep to get smoothly varying sections

closed 2022-10-01 by adam-urbanczyk