CadQuery/cadquery #530: Cannot select a hole in a chamfered object

bernhard-42, 2020-12-03:
For this example (slightly changed compared to https://github.com/CadQuery/cadquery/issues/522 to make clear circles do not have the same center)

```python
thickness = 2
height = 40
width = 65
diam = 4
tol = 0.05

l1, l2 = 50, 80
pts = [( 0,  0), ( 0, height/2), (l1, height/2 - 5), (l2, 0)]
upper_leg_hole = (l2 - 20, 0)

upper_leg = (cq.Workplane()
    .polyline(pts).mirrorX()
    .pushPoints([upper_leg_hole]).circle(diam/2 + tol).extrude(thickness)
    .edges("|Z and (not <X)").fillet(4)
)
upper_leg.faces(">Z").edges("%Circle")
```

I did not find a way to select the circle with CadQuery selectors

```python
upper_leg.faces(">Z").edges("%Circle").edges().objects

    [<cadquery.occ_impl.shapes.Edge at 0x7f24f0b94ee0>,
     <cadquery.occ_impl.shapes.Edge at 0x7f24f0b942b0>,
     <cadquery.occ_impl.shapes.Edge at 0x7f24f0cc73a0>,
     <cadquery.occ_impl.shapes.Edge at 0x7f24f0cc7ca0>]
```

[image]

`">X"` works

```python
upper_leg.faces(">Z").edges("%Circle").edges(">X").objects
  
   [<cadquery.occ_impl.shapes.Edge at 0x7f24f1be5130>]
```
[image]

`">X[0]"` fails

```python
upper_leg.faces(">Z").edges("%Circle").edges(">X[0]")

---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<ipython-input-103-be1f63569cdd> in <module>
----> 1 upper_leg.faces(">Z").edges("%Circle").edges(">X[0]")

/opt/anaconda/envs/cq-ocp21/lib/python3.8/site-packages/cadquery/cq.py in edges(self, selector, tag)
    841         returns 4, because a cube has 4 edges parallel to the z direction
    842         """
--> 843         return self._selectObjects("Edges", selector, tag)
    844 
    845     def wires(

/opt/anaconda/envs/cq-ocp21/lib/python3.8/site-packages/cadquery/cq.py in _selectObjects(self, objType, selector, tag)
    737             else:
    738                 selectorObj = selector
--> 739             toReturn = selectorObj.filter(toReturn)
    740 
    741         return self.newObject(toReturn)

/opt/anaconda/envs/cq-ocp21/lib/python3.8/site-packages/cadquery/selectors.py in filter(self, objectList)
    726         Filter give object list through th already constructed complex selector object
    727         """
--> 728         return self.mySelector.filter(objectList)

/opt/anaconda/envs/cq-ocp21/lib/python3.8/site-packages/cadquery/selectors.py in filter(self, objectList)
    605             [+\|-\|<\|>\|] \<X\|Y\|Z>
    606         """
--> 607         return self.mySelector.filter(objectList)
    608 
    609 

/opt/anaconda/envs/cq-ocp21/lib/python3.8/site-packages/cadquery/selectors.py in filter(self, objectList)
    385 
    386         # choose the Nth unique rounded distance
--> 387         nth_distance = sorted(list(objectDict.keys()), reverse=not self.directionMax)[
    388             self.N
    389         ]

IndexError: list index out of range
```

marcus7070, 2020-12-03:
I'm currently working on a PR that refactors nth selectors, I'll make sure I deal with this too. Although I haven't figured out exactly what's going wrong yet.

adam-urbanczyk, 2020-12-03:
This is the intended  - `DirectionNthSelector` only takes into account objects that are parallel (edges) or normal (faces) to the specified dir. It should be easy to add a similar selector that does not have this behavior.

marcus7070, 2020-12-03:
Whoops, I forgot that, thanks @adam-urbanczyk.

PR #549 (2021-01-05): Refactor selectors and add CenterNthSelector
This PR will close #517 and add a CenterNthSelector to resolve #530. I touched a few other things as well, sorry for lumping them all in one PR but I've at least rebased all the commits into neat chunks.

---

Commit 6d2f3ba adds _NthSelector. It is an abstract class, it should only be inherited from, not instantiated.

Previously DirectionNthSelector lumped objects at a similar distance together by rounding that distance to a certain number of decimal places. This would mean that with a tolerance of 0.1, 0.249999999 and 0.250000001 would be put int [... cut]
commit 344651d1 (2021-01-05): Added CenterNthSelector and tests

Resolves #530.
closed 2021-01-05 by jmwright