CadQuery/cadquery #531: Selector ">X[-1]" filters for perpendicular faces/edges, selector ">X" does not

marcus7070, 2020-12-04:
While looking at #530 I noticed this.

From these lines:
https://github.com/CadQuery/cadquery/blob/cf275b0ec05ded405e38936836f1aa241eef6e72/cadquery/selectors.py#L623-L628

The string selector `">X"` creates a DirectionMinMax selector, the string `">X[-1]"` creates a DirectionNthSelector.

https://github.com/CadQuery/cadquery/blob/cf275b0ec05ded405e38936836f1aa241eef6e72/cadquery/selectors.py#L399-L419

DirectionNthSelector first filters for edges/faces parallel with the direction. DirectionMinMax does not. I expected `">X"` and `">X[-1]"` result in the same behaviour. 

I would propose DirectionNthSelector should be changed to not filter for parallel. Then it could be used to select non-parallel faces and edges, such as the curves in #530. Users can always combine selectors if they need to filter for parallel first. 

DirectionMinMax could become a simple subclass of DirectionNthSelector with the index set to -1. 

This change wouldn't create any backwards compatibility issues in my code. But perhaps that is because I didn't know the difference between these selectors before now. I would like to know if it would break anything for anyone else.

Also [this page in the docs](https://cadquery.readthedocs.io/en/latest/selectors.html#filtering-edges) needs to be updated to reflect the correct classes that these string selectors create.

adam-urbanczyk, 2020-12-04:
This was intentional and is mentioned in the table here: https://cadquery.readthedocs.io/en/latest/apireference.html#selectors . The other table needs to be indeed updated.

I'm against changing this functionality, it was developed long time ago (~3 years) and meant mostly for selecting faces in complex solids (e.g. selecting the bottom of a box with a lot of fillets). For me it is a huge breaking change.

We could add another selector (or an option to this one) to support the use case from #530.

marcus7070, 2020-12-04:
Great, thanks for the feedback, IMHO this behaviour should stay then.

> We could add another selector (or an option to this one) to support the use case from #530. 

I'm working on #517 in my spare time, once I'm done with that PR I'll try doing exactly this.

I'll leave this issue open but change it to be about better docs regarding these two selectors.

commit a242473b (2021-01-05): Updated string selector doc page

Correct title underlining means the sidebar no longer shows "Combining
Selectors" as a top level item, instead it's a sub item under "String
Selectors Reference". Updated some other parts and expanded some
explanations. Resolves #531.
closed 2021-01-05 by jmwright