CadQuery/cadquery #537: exporting a wire to stl

Shimwell, 2020-12-09:
I noticed a potential bug and thought it was worth reporting.

When I make a 3D solid I can export it as a stp file and stl file, it all works fine

```python
import cadquery as cq
from cadquery import exporters

result_solid = cq.Workplane("front").lineTo(2.0, 0).lineTo(2.0, 1.0).threePointArc((1.0, 1.5),(0.0, 1.0))\
    .close().extrude(0.25)

with open('result_solid.stl', "w") as out_file:
    exporters.exportShape(result_solid, "STL", out_file)

with open('result_solid.stp', "w") as out_file:
    exporters.exportShape(result_solid, "STEP", out_file)
```
stp and stl file
[image]
[image]

However when I make a 2D wire (same as 3D but with no extrude) then the stp export works but the stl export results in a 0 byte file (hence no screen shot for that one).

```python
result_line = cq.Workplane("front").lineTo(2.0, 0).lineTo(2.0, 1.0).threePointArc((1.0, 1.5),(0.0, 1.0))\
    .close()
# note the lack of extrude here

with open('result_line.stl', "w") as out_file:
    exporters.exportShape(result_line, "STL", out_file)

with open('result_line.stp', "w") as out_file:
    exporters.exportShape(result_line, "STEP", out_file)
```
stp file
[image]

stl file is zero bytes and has no contents

I'm using the CQ master branch, Ubuntu 18.04 and python 3.8

Would be interested to know if this is a bug or I have perhaps done something wrong

adam-urbanczyk, 2020-12-09:
I think that STL can only represent triangles, it should be possible to export the face though.

Shimwell, 2020-12-09:
Ah yes that makes sense. Thanks

closed 2020-12-09 by Shimwell