CadQuery/cadquery #539: Time for 2.1 RC

adam-urbanczyk, 2020-12-10:
I propose to merge #538,  #537 and then tag 2.1RC

adam-urbanczyk, 2020-12-18:
OK, so in principle it is time for 2.1RC . What do you think about #549 ?

jmwright, 2020-12-18:
I think since it's an RC merging #549 first is fine.

adam-urbanczyk, 2020-12-18:
I was actually thinking about waiting with it. There might be still some work needed for #549 and it would be nice to at least have the RC1 before the FOSDEM deadline - I think I'll submit something after all.

adam-urbanczyk, 2020-12-18:
If you agree just tag it today.

jmwright, 2020-12-18:
> I was actually thinking about waiting with it.
> There might be still some work needed for #549 and it would be nice to at least have the RC1 before the FOSDEM deadline.
> If you agree just tag it today.

I'm not sure what you mean. As I read those three lines, the first makes me think you want to wait to do the release, but the second two seem like you want to go ahead and do it. I'm ok with either.

adam-urbanczyk, 2020-12-19:
LOL. I meant:  I was actually thinking about waiting with #549 .

jmwright, 2020-12-19:
Ok, that sounds good. Feel free to create a tag, or I'll find some time later today to do it.

jmwright, 2020-12-19:
Ok, the RC release has been tagged. https://github.com/CadQuery/cadquery/releases/tag/2.1RC1

marcus7070, 2020-12-25:
@jmwright, did you want to update changes.md?

jmwright, 2020-12-25:
> did you want to update changes.md?

Yes, that's a good point. I'll make sure it's done before the full release.

jmwright, 2020-12-29:
Changelog updates are in #557

jmwright, 2020-12-30:
I think we've completed this task, so I'm going to close.

closed 2020-12-30 by jmwright