CadQuery/cadquery #540: Incorrect cutting with filleted shapes

billingsley-john, 2020-12-10:
I have been trying to perform a cut between shapes in arranged like the following using cadquery master.

[image]

The filleted cuboids are cut from the outer round shape to create 'ports'.

When the cuboids are not filleted, the desired effect is achieved:

[image]

However, adding fillets to the cuboid edges causes problems. Sometimes cuts are not performed:

[image]

and sometimes, additionally, outer shape becomes mishapen - it is no longer has a smooth outer surface.

[image]
[image]

Does anyone know what might be causing this issue?

billingsley-john, 2020-12-10:
Apologies for the absence of a MWE. I will try to provide one soon.

billingsley-john, 2020-12-10:
Also, to note, I have checked that both the rounded outer surface and cutting cuboids are created correctly before the cut takes place in each case shown above (I have just shown the failed resulting cut).

adam-urbanczyk, 2020-12-10:
Looks like `ShapeUpgrade_UnifySameDomain` issue. Maybe try to cut with `clean=False`?

jmwright, 2020-12-11:
Those screenshots look like they could have been taken while using FreeCAD. If you're using FreeCAD (CadQuery 1.x). I'd like to confirm whether you're using the newer 2.x version of FreeCAD. The following code works fine for me in the latest CQ-editor binary build.
```
import cadquery as cq

# A very crude approximation of your sphere
core = cq.Workplane("XY").sphere(10, angle3=180).cut(cq.Workplane("XY").sphere(9.8)).center(0, 0).circle(3.0).cutThruAll()

# The cutout object
cutout = cq.Workplane("YZ").rect(3, 3).extrude(13).edges("|X").fillet(0.5).rotateAboutCenter((0, 0, 1), 45)#.transformed(rotate=(0, 0, 45))

result = core.cut(cutout)

show_object(result)
```
[image]

Interestingly though if I use the "ZX" workplane for the cutout, the fillet is lost.

[image]

jmwright, 2020-12-11:
I forgot to try the cut with `clean=False`. It does not fix the issue when using the ZX plane.
```python
import cadquery as cq

# A very crude approximation of your sphere
core = cq.Workplane("XY").sphere(10, angle3=180).cut(cq.Workplane("XY").sphere(9.8)).center(0, 0).circle(3.0).cutThruAll()

# The cutout object
cutout = cq.Workplane("ZX").rect(3, 3).extrude(13).edges("|X").fillet(0.5).rotateAboutCenter((0, 0, 1), 45)#.transformed(rotate=(0, 0, 45))

result = core.cut(cutout, clean=False)

show_object(result)
```

adam-urbanczyk, 2020-12-11:
This is what I see for ZX:
[image]

jmwright, 2020-12-11:
Ah, I didn't think about that happening. I'm assuming the ends are filleted because what we see as the green Y axis becomes the `xDir` (0, 1, 0) for the workplane when `ZX` is used.

jmwright, 2020-12-11:
Oh, and another thing I made a mistake on is rotating the cutout bar about the center. In the original screenshots that bar is rotated so that it's aligned with the center of the sphere. Using `rotate()` instead with the appropriate axis vectors should fix that.

jmwright, 2020-12-11:
I guess I'm confused about the ZX plane after looking at the docs. https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=plane#cadquery.Plane

billingsley-john, 2020-12-11:
Hi @adam-urbanczyk and @jmwright , thank you both for your responses. I just wanted to mention a few things that might help solve this issue!

Unfortunately, performing cut with `clean=False` didn't solve the problem, however, the result produced wasn't identical. As pictured below, with clean=False the partial outline of a missing ports became visible.

[image]

Another thing is that the multiple cutting cuboids are not a set of individual solids - together they are one solid. I don't know whether this changes your outlook on the issue.

Finally, I am only using FreeCAD to view the geometry. I use CadQuery 2 (master branch) to create the solids and export them as stp files, but view them using FreeCAD.

Thanks again, and I will be sure to update this issue with any further developments!

adam-urbanczyk, 2020-12-13:
@billingsley-john  could you export the two shapes before cutting as breps and share them here? This is the simplest way to repoduce the issue.

billingsley-john, 2020-12-17:
Hi @adam-urbanczyk, sorry for the delay. I have the breps of the shapes. Do you just want the contents of each file in a comment on this issue? I can't share the file itself.

adam-urbanczyk, 2020-12-17:
You can zip the brep files and attach in a comment.

billingsley-john, 2020-12-18:
[geometry.zip](https://github.com/CadQuery/cadquery/files/5714618/geometry.zip)

adam-urbanczyk, 2021-03-21:
@billingsley-john I tried different cut settings using OCCT directly, but did not manage to get any improvement.

Assuming that your spherical object is made using `parametricCurve` could you check if #694 has any effect?

billingsley-john, 2021-03-30:
Thank you for the response @adam-urbanczyk. Unfortunately, #694 did not fix this specific problem - construction errors were still observed when cuts were made with filleted shapes. The cutting shape is created by filleting the edges of cadquery solid, which is created by extruding a polyline connecting points together.