CadQuery/cadquery #544: construction error with 360 degree circle

Shimwell, 2020-12-13:
I was just trying to union a couple of shapes and got it working when the shapes are revolved by 180 but appear to get an error when revolving to 360 degrees.

I've made a minimal example here

```python
import cadquery as cq
from cadquery import exporters

angle = 360 # 360 causes error by 180 works

triangle = (
    cq.Workplane("XY")
    .polyline([
        (4235.0, 4170.0),
        (2862.237128095924, 2191.6819397473832),
        (3200.5113335386786, 1995.5912415180355),
    ])
    .close()
    .revolve(angle)
)

circle = (
    cq.Workplane("front")
    .moveTo(3700., 3100.)
    .circle(250)
    .revolve(angle)
)

both = circle.union(triangle)

with open('both.stp', "w") as f:
    exporters.exportShape(both, "STEP", f)
```

There error that occurs appears to be caused by the union of the two shapes when they are 360 revolves. The union works for a 180 degree rotation.

[image]

This is the shape when using ```angle=180``` exported to a stp file using CadQuery 2 (master version). I just used FreeCAD for the loading of the step file. The python code was run with the latest CQ master version.

[image]

jmwright, 2020-12-13:
@Shimwell If you use a revolution angle of 359 or 359.9, does it still work like the 180 degree revolve? We've seen that problem before in a somewhat different context.

adam-urbanczyk, 2020-12-13:
It's not a revolve issue - it is related to the final union. Traditionally `clean=False` does the job
```python
angle = 360 # 360 causes error by 180 works

triangle = (
    cq.Workplane("XY")
    .polyline([
        (4235.0, 4170.0),
        (2862.237128095924, 2191.6819397473832),
        (3200.5113335386786, 1995.5912415180355),
    ])
    .close()
    .revolve(angle)
)

circle = (
    cq.Workplane("front")
    .moveTo(3700., 3100.)
    .circle(250)
    .revolve(angle)
)

both = circle.union(triangle, clean=False)
```

Shimwell, 2020-12-13:
Thanks for the solutions.

I can confirm that the setting the revolved to 359 and 359.9 the shape with the union worked over here.

Also the clean = False arg in the union allowed 360 revolutions which is just perfect for my use case.

Many thanks

closed 2020-12-13 by Shimwell

adam-urbanczyk, 2020-12-13:
What do you mean, I can get the following result that looks IMO fine (section added for clarity):
[image]
```python
both = circle.union(triangle, clean=False).shell(-50)
bot_sh = both.section()
```

Shimwell, 2020-12-13:
ah it is working again now, I'm sure it was causing difficulty. Let me figure this out a bit more I think there is one more thing to solve but my script involves so many stages I need to narrow it done a bit more.

Shimwell, 2020-12-13:
[image]

Shimwell, 2020-12-14:
Everything is working and the clean=False is happily working with the shell. I was trying to shell a shape that wasn't completed revolved and the spline edges were not working that well. But revolving it 360 degrees so that there were less faces made it easier to shell.