CadQuery/cadquery #546: How to extrude a polyline without mirror?

nopria, 2020-12-13:
In the documentation there is [an example of how extrude a polyline](https://cadquery.readthedocs.io/en/latest/examples.html#polylines), but if I don't mirror the polyline, either closed or not, I get a "list index out of range" error:

```python
import cadquery as cq

(L,H,W,t) = ( 100.0, 20.0, 20.0, 1.0)
pts = [
    (0,H/2.0),
    (W/2.0,H/2.0),
    (W/2.0,(H/2.0 - t)),
    (t/2.0,(H/2.0-t)),
    (t/2.0,(t - H/2.0)),
    (W/2.0,(t -H/2.0)),
    (W/2.0,H/-2.0),
    (0,H/-2.0)
]
result = cq.Workplane("front").polyline(pts).extrude(L)
```

Bug or not, this seems pretty counter-intuitive to me. What should I do to extrude a simple (not mirrored) polyline?

jmwright, 2020-12-13:
Try calling `close()` before the extrude.

nopria, 2020-12-13:
That resolves, thanks. Since there are no examples of closed polylines, I suggest adding to that example a simple notice like "Please note that when extruding a generic non mirrored polyline you must explicitly call `close()` before extruding."

closed 2020-12-13 by nopria

adam-urbanczyk, 2020-12-13:
Actually what you need is conversion from edges to wire (`wire()`).

nopria, 2020-12-14:
Actually I needed a closed polyline, so `close()` is enough for me.

From some tests it seems to me that after mirror, regardless of polyline being closed or not, extruding is possible. Since a non-mirrored polyline must be closed to be extruded, isn't that confusing? What is the reasoning behind this?

dcowden, 2020-12-14:
@nopria to extrude you need a closed curve. Mirroring an open curve frequently produces a closed one, so closing it is not needed. 

It is possible that a curve is not closed after it is mirrored... In that case extrusion would fail to produce a solid also

nopria, 2020-12-14:
Are you sure that extrusion fail on a mirrored open curve? I'm able to extrude an open curve with

```python
cq.Workplane("YZ").polyline([(0,0),(10,0),(10,10)]).mirrorY().extrude(100) # extrude working on open curve
```

while I'm not able to extrude a curve closed "by hand", which I would expect to behave like curves closed by mirroring (so without explicit `close()` command):

```python
cq.Workplane("YZ").polyline([(0,0),(10,0),(10,10),(0,10),(0,0)]).extrude(100) # extrude failing on curves closed "by hand"
```

That's counter-intuitive IMHO.

adam-urbanczyk, 2020-12-14:
As mentioned above, it is due to conversion from `Edges` to `Wire` using for example the `wire()` method. This works fine:
```python
r = cq.Workplane("YZ").polyline([(0,0),(10,0),(10,10),(0,10),(0,0)]).wire().extrude(100)
```