CadQuery/cadquery #548: Rotated filled arc

danaki, 2020-12-15:
This seems renders incorrectly:
```
result = cq.Workplane("XZ") \
    .moveTo(0, 0) \
    .ellipseArc(4, 4, 0, 60, startAtCurrent=False) \
    .lineTo(0, 0) \
    .ellipseArc(4, 4, 120, 180, startAtCurrent=False) \
    .lineTo(0, 0) \
    .ellipseArc(4, 4, 240, 300, startAtCurrent=False) \
    .lineTo(0, 0) \
    .close() \
    .extrude(0.5)
```

This is the result I want:
```
solid0 = cq.Workplane("XZ") \
    .moveTo(0, 0) \
    .ellipseArc(4, 4, 0, 60, startAtCurrent=False) \
    .lineTo(0, 0) \
    .close() \
    .extrude(0.5)

solid1 = solid0.rotate((0, -1, 0), (0, 1, 0), 120)
solid2 = solid0.rotate((0, -1, 0), (0, 1, 0), 240)

solid = cq.Workplane("XZ")
solid = solid.union(solid1)
solid = solid.union(solid2)

result = solid
```

Is it a bug or I'm doing something wrong?

adam-urbanczyk, 2020-12-16:
You did not close the wires, this works as expected:
```python
result = (
    cq.Workplane("XZ")
    .moveTo(0, 0)
    .ellipseArc(4, 4, 0, 60, startAtCurrent=False)
    .lineTo(0, 0)
    .close()
    .moveTo(0,0)
    .ellipseArc(4, 4, 120, 180, startAtCurrent=False)
    .lineTo(0, 0)
    .close()
    .moveTo(0,0)
    .ellipseArc(4, 4, 240, 300, startAtCurrent=False)
    .lineTo(0, 0)
    .close()
    .extrude(1)
    )
```

I'd personally extrude one element and then use `polarArray`.

danaki, 2020-12-16:
Yes, I tried polarArray but it also didnt work. Thank you for the explanation.

closed 2020-12-16 by danaki

danaki, 2020-12-16:
Can't make it work using polarArray