CadQuery/cadquery #553: How to select objects that are not parallel to any of the dimensions

rowanG077, 2020-12-24:
I have a simple design where I want to select an edge that is at an angle. But no matter what I try I cannot get the edge selected.

This is my part:

```Python
from __future__ import division
from cadquery import *

stand_clr = 0.4
stand_height = 90
stand_width = 294.7
stand_thickness =  1
stand_supp_thickness =  1
stand_supp_width = 12.5

stand_lip_chamfer_height = 0.5
stand_lip_chamfer_depth = 2.5 * stand_lip_chamfer_height
    
stand = cq.Workplane("front")\
    .vLine(stand_thickness + stand_supp_thickness - stand_lip_chamfer_height)\
    .line(stand_lip_chamfer_depth, stand_lip_chamfer_height)\
    .hLine(stand_supp_width - stand_supp_thickness - stand_lip_chamfer_depth)\
    .line(stand_supp_thickness, -stand_supp_thickness)\
    .hLine(stand_width - 2 * stand_supp_width)\
    .line(stand_supp_thickness, stand_supp_thickness)\
    .hLine(stand_supp_width - stand_supp_thickness - stand_lip_chamfer_depth)\
    .line(stand_lip_chamfer_depth, -stand_lip_chamfer_height)\
    .vLine(-stand_thickness - stand_supp_thickness + stand_lip_chamfer_height)\
    .close().extrude(-stand_height)

show_object(stand)
```

And and I want to add a fillet to the following edge on both sides:

[image]

I have tried many things. First selecting faces then the edges. Trying combining selectors etc. But I can't even get the face that is at 45 degrees selected.

marcus7070, 2020-12-25:
It was awkward but I got this to work:
```python
from __future__ import division
from cadquery import *

stand_height = 50
stand_width = 50
stand_thickness =  1
stand_supp_thickness =  1
stand_supp_width = 12.5

stand_lip_chamfer_height = 0.5
stand_lip_chamfer_depth = 2.5 * stand_lip_chamfer_height
    
stand = cq.Workplane("front")\
    .vLine(stand_thickness + stand_supp_thickness - stand_lip_chamfer_height)\
    .line(stand_lip_chamfer_depth, stand_lip_chamfer_height)\
    .hLine(stand_supp_width - stand_supp_thickness - stand_lip_chamfer_depth)\
    .line(stand_supp_thickness, -stand_supp_thickness)\
    .hLine(stand_width - 2 * stand_supp_width)\
    .line(stand_supp_thickness, stand_supp_thickness)\
    .hLine(stand_supp_width - stand_supp_thickness - stand_lip_chamfer_depth)\
    .line(stand_lip_chamfer_depth, -stand_lip_chamfer_height)\
    .vLine(-stand_thickness - stand_supp_thickness + stand_lip_chamfer_height)\
    .close().extrude(-stand_height)

show_object(stand, "stand")

sel = cq.selectors.SumSelector(
    cq.selectors.NearestToPointSelector((stand_supp_width, 0, stand_thickness + stand_supp_thickness / 2)),
    cq.selectors.NearestToPointSelector((stand_width - stand_supp_width, 0, stand_thickness + stand_supp_thickness / 2)),
)

fstand = (
    stand
    .faces(">Z")
    .edges(sel)
    .fillet(1)
)

show_object(fstand, "filleted stand")
```

[image]

I feel like there should be some combination of selectors and InverseSelector that could have done this as well, but I couldn't get there.

adam-urbanczyk, 2020-12-25:
You are probably looking for this:
```python
stand.faces('>Z').edges('|(1,1,0) or |(1,-1,0)')
```

naggie, 2020-12-26:
> You are probably looking for this:
> 
> ```python
> stand.faces('>Z').edges('|(1,1,0) or |(1,-1,0)')
> ```

Wow, that's neat. I was wondering how to select faces that might be otherwise unreachable. Thanks!