CadQuery/cadquery #562: Expose BRepProj_Projection

adam-urbanczyk, 2021-01-05:
Expose `BRepProj_Projection` on the occ_impl level. Prerequisite of #361 .

https://old.opencascade.com/doc/occt-7.5.0/refman/html/class_b_rep_proj___projection.html

gumyr, 2022-01-14:
Here is my implementation for projection of Edge, Wire and Face:
```python

def _projectWireToShape(
    self: Union[cq.Wire, cq.Edge],
    targetObject: cq.Shape,
    direction: cq.Vector = None,
    center: cq.Vector = None,
) -> tuple[list[cq.Wire]]:
    """
    Project a Wire onto a Solid generating new Wires on the front and back of the object
    one and only one of `direction` or `center` must be provided

    To avoid flipping the normal of a face built with the projected wire the orientation
    of the output wires are forced to be the same as self.
    """
    if not (direction is None) ^ (center is None):
        raise ValueError("One of either direction or center must be provided")

    if not direction is None:
        projection_object = BRepProj_Projection(
            self.wrapped,
            cq.Shape.cast(targetObject.wrapped).wrapped,
            gp_Dir(*direction.toTuple()),
        )
    else:
        projection_object = BRepProj_Projection(
            self.wrapped,
            cq.Shape.cast(targetObject.wrapped).wrapped,
            gp_Pnt(*center.toTuple()),
        )

    target_orientation = self.wrapped.Orientation()

    # Generate a list of the projected wires with aligned orientation
    output_wires = []
    while projection_object.More():
        projected_wire = projection_object.Current()
        if target_orientation == projected_wire.Orientation():
            output_wires.append(cq.Wire(projected_wire))
        else:
            output_wires.append(cq.Wire(projected_wire.Reversed()))
        projection_object.Next()

    # BRepProj_Projection is inconsistent in the order that it returns projected
    # wires, sometimes front first and sometimes back - so sort this out
    front_wires = []
    back_wires = []
    if len(output_wires) > 1:
        output_wires_centers = [w.Center() for w in output_wires]
        projection_center = reduce(
            lambda v0, v1: v0 + v1, output_wires_centers, cq.Vector(0, 0, 0)
        ) * (1.0 / len(output_wires_centers))
        output_wires_directions = [
            (w - projection_center).normalized() for w in output_wires_centers
        ]
        if not direction is None:
            direction_normalized = direction.normalized()
        else:
            direction_normalized = (center - projection_center).normalized()
        for i, d in enumerate(output_wires_directions):
            # If wire direction from center of projection aligns with direction
            # it's considered a "front" wire
            if d.dot(direction_normalized) > 0:
                front_wires.append(output_wires[i])
            else:
                back_wires.append(output_wires[i])
    else:
        front_wires = output_wires
    return (front_wires, back_wires)
[... 176 lines cut ...]
        except StdFail_NotDone as e:
            raise ValueError(
                "Error adding interior hole in non-planar face with provided interiorWires"
            ) from e
        if not surface_face.isValid():
            raise ValueError("Invalid non-planar face from provided interiorWires")

    return surface_face

def _makeNonPlanarFace(
    self,
    surfacePoints: list[cq.Vector] = None,
    interiorWires: list[cq.Wire] = None,
) -> cq.Face:
    return __makeNonPlanarFace(self, surfacePoints, interiorWires)

cq.Wire.makeNonPlanarFace = _makeNonPlanarFace
```
This method adds interior holes (and corrects the non cadquery input types) to `makeNSidedSurface`. I've found the set of try/except blocks to be extremely useful. By giving the user different exceptions for external/internal wires and surface points it's possible to debug face creation. Without it I found it essentially impossible to debug complex faces with outer and inner wires with surface points.

There is quite a lot of code here but I haven't found a less complex way to achieve these results.

Cheers,
Roger
P.S. Any PR would have VectorLike inputs - not Vector as shown here.

bragostin, 2022-02-16:
I think this is great, maybe as a plugin?

gumyr, 2022-02-16:
> This is a CQ issue, so it is planned to be in CQ. I'd prefer that this
> (and related issue - e.g. offset) is implemented by the core team

In the meantime, you can use my versions here:
https://cq-warehouse.readthedocs.io/en/latest/extensions.html#extensions_doc.Face.projectToShape
(along with many other cq_warehouse features).

Cheers,
Roger

closed 2022-07-08 by adam-urbanczyk