CadQuery/cadquery #565: Topology traversal and topology based selectors

adam-urbanczyk, 2021-01-06:
E.g. find all wires connected to a vertex

Look here for reference: https://github.com/tpaviot/pythonocc-utils/blob/master/OCCUtils/Topology.py

michaelgale, 2021-01-07:
In a way, I use custom Selectors (in cqkit) to achieve this result, i.e. topology traversal.  Using the example of finding all wires connected to a vertex:

```python
import cad query as cq
from cqkit import *

# make a cube
r = cq.Workplane("XY").rect(1, 2).extrude(5)
# make a selector based on a vertex on the top corner of cube
vs = SharedVerticesWithObjectSelector(obj=Vertex.makeVertex(0.5, 1.0, 5.0))
# select wires connected to vertex
w = r.wires(vs).vals()
# w contains the three wires connected to vertex as expected
```

The `Selector` classes are easily CQ's best feature.  I can't imagine being able to build some of my complex models using other traditional CAD tools.  The Selectors with their pythonic arithmetic, logical and programmatic composition allow for very powerful topology navigation and selection.

michaelgale, 2021-01-07:
As a followup to the above example, you can just as easily reuse the selector to find all shared shape objects:
```python
# select faces common with vertex
f = r.faces(vs).vals()
# and edges
e = r.edges(vs).vals()

# create another selector for the diagonally opposite vertex
vs2 = SharedVerticesWithObjectSelector(obj=Vertex.makeVertex(-0.5, -1.0, 5.0))
# now select the single wire at the top
wtop = r.wires(AndSelector(vs, vs2)).vals()
# or all wires except bottom "floor"
w = r.wires(vs + vs).vals()
```