CadQuery/cadquery #566: CQ is going to FOSDEM2021

adam-urbanczyk, 2021-01-06:
Location/devroom: [open source computer aided modeling and design](https://fosdem.org/2021/schedule/track/open_source_computer_aided_modeling_and_design)

If you have some nice models/screenshots to share please do so here!

jmwright, 2021-01-06:
I think I've previously mentioned a side project of mine involving a tensegrity structure. There is still a bug in the string placement code so a couple of those are out of place, but I can have it fixed pretty quickly to give a better screenshot if you're interested in using it. If you don't want to use it, no problem.

It's based on the mathematics laid out in a research paper, and I implemented it (with a few shortcuts) in CadQuery via Numpy, and then used CQ's newer assembly functionality to put everything together. In theory, I can change the major and minor radii of the torus and have everything adjust appropriately.

Blue = bar
Red = String
[image]

michaelgale, 2021-01-07:
As always, feel free to use any media here:  https://github.com/fx-bricks/fx-cad-notes

Also, you can find a high quality render(with Blender from STL exported from CQ) of one of our newest CQ built products here:
https://www.flickr.com/photos/michaelgale/50024337707/in/dateposted-public/

and the real thing soon to be released on the market here:
https://www.flickr.com/photos/michaelgale/50674016407/in/dateposted-public/

adam-urbanczyk, 2021-01-07:
Both examples look great to me, thanks!

bragostin, 2021-01-07:
How about this one? The Tetrakaidecahedron (Kelvin Cell) can pave 3D space and is often encountered in cristallography.
[image]

jmwright, 2021-01-07:
@bragostin Very cool. Do you have a link to some code?

@adam-urbanczyk I think the router spindle assembly would be a good one too.

bragostin, 2021-01-07:
@jmwright  yes, there https://github.com/bragostin/CadQuery-Examples/blob/main/Tetrakaidecahedron.py
It can surely be improved.

jmwright, 2021-01-10:
@adam-urbanczyk If you decide to include it, tensegrity structure image is [here](https://github.com/jmwright/IslandOne/blob/master/docs/static/images/i1_full_tensegrity_structure_with_ground_deck.png) and the readme with background info is [here](https://github.com/jmwright/IslandOne/blob/master/README.md).

adam-urbanczyk, 2021-02-19:
Thanks everyone, the video is online:
https://video.fosdem.org/2021/D.cad/cadquery.webm

closed 2021-02-19 by adam-urbanczyk