CadQuery/cadquery #567: Same options for centered in methods .rect and .box

marcus7070, 2021-01-07:
I think we can change the way we specify `centered` in both `.rect` and `.box` to be more consistent and more useful. Basically: 
* Keep existing behaviour for `.rect(1, 1, centered=True)` and `.box(1, 1, 1, centered=(True, True, True)`
* `.rect(10, 10, centered=(True, False)` produces a rectangle with corners on (-5, 0) and (5, 10)
* `.box(1, 1, 1, centered=True` is the same as `.box(1, 1, 1, centered=(True, True, True)`

Thoughts?

marcus7070, 2021-01-07:
These methods also use a 2 or 3d centered option, and could be given the same treatment:

* `rarray`
* `sphere`
* `wedge`

RubenRubens, 2021-01-07:
I think it's a good idea. Making stuff more consistent is always welcome.

michaelgale, 2021-01-07:
I like this suggestion.  I can't count how many times I adjust objects by `(-xyz/2)`!  As long as the default behaviour isn't broken, this would be useful.

adam-urbanczyk, 2021-01-07:
Sounds good, do you mind waiting till after 2.1?

marcus7070, 2021-01-07:
> Sounds good, do you mind waiting till after 2.1?

No worries, that is fine.

PR #617 (2021-02-10): Centered option
Will resolve #567. 

Also fixes some errors in the HTML docs for box and wedge.
closed 2021-02-10 by adam-urbanczyk