CadQuery/cadquery #570: Lofting wires created with offset2D() can result in wrong offset size

tanius, 2021-01-11:
Lofting does not work as expected when creating two identical wires starting from the same workplane object, if these wires use `offset2D()`. In that case, the first wire created will be too large by a constant measure, as if an additional measure would be added to its `offset2D()` operation somehow.

As a workaround, I was able to use a [special technique](https://github.com/CadQuery/cadquery/issues/327#issuecomment-616127686) where the wires are created independently at first.

The following screenshot demonstrates the issue (red object) and the expected result using the workaround (green object). Both objects should be identical, but are not. When removing the `.offset2D(5, "arc")` operation from the profile, red and green shape become identical, so it's only the combination of `offset2D()` and `loft()` that creates this issue.

[image]

The code that produced the above rendering (as copied from CQ-editor):

```
def myprofile(self):
    return self.hLine(10).vLine(10).hLine(-10).close().offset2D(5, "arc")

def good_part(h = 5):
    part = cq.Workplane("XY")
    part.ctx.pendingWires.extend(
        cq.Workplane("XY").myprofile().ctx.pendingWires
    )
    part.ctx.pendingWires.extend(
        cq.Workplane("XY").transformed(offset = (0, 0, h)).myprofile().ctx.pendingWires
    )
    return part.loft(combine = True)

def bad_part(h = 5):
    return (
        cq
        .Workplane("XY").myprofile()
        .workplane(offset = h).myprofile()
        .loft(combine = True)
    )

cq.Workplane.myprofile = myprofile

good_part = good_part()
show_object(good_part, name = "good part", options = {"color": "green", "alpha": 0})

bad_part = bad_part()
show_object(bad_part, name = "bad part", options = {"color": "red", "alpha": 0.8})
```

I'm using CadQuery and CQ-editor installed from their master branches two days ago, using Conda.

tanius, 2021-01-11:
As an even simpler example, I just notice it also works with offset circles, using this `profile()` method with the above code:

```
def myprofile(self): return self.circle(5).offset2D(5, "arc")
```

So it's not an issue with straight lines, but with `offset2D()` in more general for all kinds of 2D shapes.

marcus7070, 2021-01-11:
Basically you're offsetting the first wire twice in your `bad_part` method. Remove the first `offset2D` and you'll get the result you want.

If you look at the source:

https://github.com/CadQuery/cadquery/blob/f4b5e8210b16875a298e3461dd3fa8b9e5e9ec68/cadquery/cq.py#L3695-L3713

`offset2D` gets it's Shapes from `_consolidateWires`.

https://github.com/CadQuery/cadquery/blob/f4b5e8210b16875a298e3461dd3fa8b9e5e9ec68/cadquery/cq.py#L2047-L2056

`_consolidateWires` takes objects from `ctx.pendingEdges` and `pendingWires`. 

So basically the `offset2D` method takes `ctx.pendingWires` -> offset -> back into `pendingWires`.

tanius, 2021-01-13:
Thank you very much for the detailed explanations! Closing now, as it's not a bug after all.

I'll certainly consult the source next time I am stuck. It's much more compact and less intimidating than I had expected. :slightly_smiling_face:

closed 2021-01-13 by tanius