CadQuery/cadquery #572: Helix example is no longer working

RubenRubens, 2021-01-12:
This code worked one year ago, it can be found on issue #271. Running the same code today causes `ValueError: Wire or Edge instance required` from the `_sweep` method.

```python
import cadquery as cq

r = 3 # Radius of the helix
p = 1.5 # Pitch of the helix
h = 4 # Height of the helix

# Helix
wire = cq.Wire.makeHelix(pitch=p, height=h, radius=r)
shape = cq.Wire.combine([wire])
helix = cq.Workplane("XY").newObject([shape])

# Final result. A circle sweeped along a helix.
result = (
    cq.Workplane('XZ')
    .center(r, 0)
    .circle(0.1)
    .sweep(helix, isFrenet=True)
)
```

_Tested with the latest CQ version (build with conda from master)_

adam-urbanczyk, 2021-01-12:
`Wire.combine` was fixed during the annotation process. So you need to change the code to:

```python
import cadquery as cq

r = 3 # Radius of the helix
p = 1.5 # Pitch of the helix
h = 4 # Height of the helix

# Helix
wire = cq.Wire.makeHelix(pitch=p, height=h, radius=r)
shape = cq.Wire.combine([wire])
helix = cq.Workplane("XY").newObject(shape)

# Final result. A circle sweeped along a helix.
result = (
    cq.Workplane('XZ')
    .center(r, 0)
    .circle(0.1)
    .sweep(helix, isFrenet=True)
)
```

BTW: the combine call is not needed at all. You can also use:
```python
import cadquery as cq

r = 3 # Radius of the helix
p = 1.5 # Pitch of the helix
h = 4 # Height of the helix

# Helix
wire = cq.Wire.makeHelix(pitch=p, height=h, radius=r)
helix = cq.Workplane(obj=wire)
# Final result. A circle sweeped along a helix.
result = (
    cq.Workplane('XZ')
    .center(r, 0)
    .circle(0.1)
    .sweep(helix, isFrenet=True)
)
```

RubenRubens, 2021-01-12:
Thanks! This is really helpful.

PR #573 (2021-01-15): Indicate additional breaking changes + docstring fixes
Will resolve #572
closed 2021-01-15 by adam-urbanczyk