CadQuery/cadquery #576: Cannot draw array on a 2-dimensional plane

brianng0305, 2021-01-13:
I have no problem with using below code on 3d array(add .extrude). But only 1 circle appears when I would like to draw a 10*2 array on a 2-dimensional plane. What is the problem?

`import cadquery as cq

result = cq.Workplane("XY").rarray(100,100,10,2).circle(10)

show_object(result)`
Thank you for your help.

[image]

jmwright, 2021-01-13:
@brianng0305 Unless you have a compelling reason to run the FreeCAD workbench, I would suggest switching to CQ-editor. The FreeCAD workbench does not support CadQuery 2.x. There are still old tutorial videos pointing to FreeCAD, but it is non-trivial to make it compatible with CadQuery 2.x. CQ-editor repo: https://github.com/CadQuery/CQ-editor

Your code works fine for me in CQ-editor.
[image]

adam-urbanczyk, 2021-01-13:
I can also confirm that it works on the latest CQ/CQ-editor.

brianng0305, 2021-01-14:
Thank you jmwright and Adam. It works under CQ-editor.

I didn't use CQ editor since it crushed to the existing conda-environment. It works with the new conda-environment.

Thanks!!

closed 2021-01-14 by brianng0305