CadQuery/cadquery #577: moving a workplane origin

billingsley-john, 2021-01-13:
Hi, I have been using cadquery to produce swept solids. To position the 'end faces' of the solid at specified positions I have been using code similar to this:

`solid = cq.Workplane("XZ").moveTo(50, 0).workplane()`
`solid = solid.polyline([(-10, 10), (10, 10), (10, -10), (-10, -10)]).close()`

Previously, this would create a 20x20 square centered on (50, 0). However, this doesn't work anymore, the square now centers on (0, 0).

Could this be a bug, or maybe this is now the wrong approach to take?

My understanding was that `.moveTo(50, 0).workplane()` would essentially create a new origin at (50, 0) so that the polyline would be placed correctly.

I also notice that if .circle() is used instead of .polyline(), a circle is correctly centered at (50, 0).

Any help with this problem would be greatly appreciated.

Thanks,

John

jmwright, 2021-01-13:
@billingsley-john Are you running on the latest master of cadquery? I can't replicate your problem with the latest binary development release of CQ-editor which may be a little bit behind master.

[image]

You could also try moving the center of the workplane. `moveTo` is more of a 2D drawing method if I remember correctly.

You can try this, which also works for me in CQ-editor.
```python
solid = cq.Workplane("XZ").center(50, 0).workplane()
solid = solid.polyline([(-10, 10), (10, 10), (10, -10), (-10, -10)]).close()
```

billingsley-john, 2021-01-13:
Hi @jmwright,

Yes, I am using the latest cq master (installed today), and your suggestion of using center() worked great! Not too sure what's going on locally for me with moveTo() but at least there is a viable alternative.

Thank you very much!

adam-urbanczyk, 2021-01-13:
This is due to #532 (solving #314). To get the olde behaviour you nee to use `centerOption="CenterOfMass"``.
```python
solid = cq.Workplane("XZ").moveTo(50, 0).workplane(centerOption="CenterOfMass")
solid = solid.polyline([(-10, 10), (10, 10), (10, -10), (-10, -10)]).close()
```

marcus7070, 2021-01-14:
> ```python
> solid = cq.Workplane("XZ").center(50, 0).workplane()
> solid = solid.polyline([(-10, 10), (10, 10), (10, -10), (-10, -10)]).close()
> ```

The `.workplane()` isn't needed there, `.center(x, y)` creates a new plane already.

billingsley-john, 2021-01-14:
Great, thank you all for your responses!