CadQuery/cadquery #580: Question about BoundingBox

bragostin, 2021-01-14:
I would like to retrieve the bounding box of a created shape in terms of [xmin, ymin, zmin, xmax, ymax, zmax].
I found this function https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=boundingbox#cadquery.Shape.BoundingBox
but it does not seems to do that, it returns an object.
Is there an easy way to get the [xmin, ymin, zmin, xmax, ymax, zmax] bounding box of an object?

jmwright, 2021-01-14:
The [BoundingBox class in geom.py](https://github.com/CadQuery/cadquery/blob/5b5d8e0572349877942bb378e26696cc3c563bf4/cadquery/occ_impl/geom.py#L729) has those values. You use it like this:

```python
import cadquery as cq

bb = cq.Workplane().box(10, 10, 10).val().BoundingBox()
log(bb.xmin)
log(bb.xmax)
# etc...
```

bragostin, 2021-01-14:
Neat! Thank you!

closed 2021-01-14 by bragostin