CadQuery/cadquery #583: Workplane.clean() function doesn't work

jongra, 2021-01-15:
I've installed the current master version of `cadquery` and `cq-editor` via miniconda3 (python 3.8).
Both run mostly without errors.

Only the` clean()` function is not rendered and objects do not vanish from the rendered view. It raises no error.
The same behaviour happens for an indirect call of `clean()`, for example through `cut()`.

I've appended a screen shot of the "Parametric Enclosure" example.
[image]

adam-urbanczyk, 2021-01-15:
`Workplane.clean()` does something else than you think: 

https://github.com/CadQuery/cadquery/blob/3098c90080c77bd057b07615d46e8d1d5feeedb2/cadquery/cq.py#L3564

jongra, 2021-01-15:
@adam-urbanczyk - ouh, you're right, excuse my wrong conclusion and thank you a lot for your immediate reaction to my question.
I don't know where I got this wrong, but I thought that `clean()` would remove the object, and that the bool parameter `clean` of the `cut()` function sets if a object is removed after cutting. This is as you mentioned not the case.

Anyways, the `cut()` function should cut the parameter solid from the current solid, or am I wrong again?
As it is visible in the screen shot, this is not the case for my program.
If you could confirm, that this is indeed an error, I'll reformulate my question.

adam-urbanczyk, 2021-01-16:
There are multiple objects displayed, show the final object with `show_object` and then check if all worked fine.

jongra, 2021-01-18:
Yes, the problem was, that it displayed all objects. 
When I select single objects the result is as expected. All works fine.
Thank you for your help!

closed 2021-01-18 by jongra