CadQuery/cadquery #586: Array an object

brianng0305, 2021-01-17:
Good morning.

I would like to make an array of L shape

so I type:
`L = cq.Workplane("XY").circle(10).extrude(-500)
L = L.transformed(offset=cq.Vector(0,0,-500)).transformed(rotate=cq.Vector(90,0,0))\
.circle(12).extrude(-500)` 

and then
`Ls = cq.Workplane("XY").rarray(100, 100, 10, 1, center=True).L`

Cq-editor said workplane has no attribute L.

So i type:
`Ls = cq.Workplane("XY").rarray(100, 100, 10, 1, center=True).circle(12).extrude(-500).transformed(offset=cq.Vector(0,0,-500)).transformed(rotate=cq.Vector(90,0,0)).rarray(100, 100, 10, 1, center=True).circle(12).extrude(-500)
`
It works.

[image]

I tried to use for loop to simplfy the procedure of array. Cq-editor said int can not be iterable. I failed.:

`n = 5
for x in 20:
    if x % n == 0:
L = cq.Workplane("XY").transformed(offset=cq.Vector(x,0,0).circle(10).extrude(-500)
L = L.transformed(offset=cq.Vector(0,0,-500)).transformed(rotate=cq.Vector(90,0,0))\
.circle(12).extrude(-500)
`
Can I simplfy the procedure of array objects?

Thank you.

jmwright, 2021-01-17:
You could maybe use CadQuery's plugin system. https://cadquery.readthedocs.io/en/latest/extending.html#extending-cadquery-plugins
```python
import cadquery as cq

def createLs(self):
    L = cq.Workplane().circle(10).extrude(-500)
    L = L.transformed(offset=cq.Vector(0,0,-500)).transformed(rotate=cq.Vector(90,0,0)).circle(12).extrude(-500)

    return self.eachpoint(lambda loc: L.val().located(loc), True)

cq.Workplane.createLs = createLs

L2 = cq.Workplane("XY").pushPoints([(0, 0), (100, 0), (200, 0)]).createLs()

show_object(L2)
```
[image]

brianng0305, 2021-01-18:
thank you very much.

closed 2021-01-18 by brianng0305