CadQuery/cadquery #587: Sweep along 3D curve???

daemacles, 2021-01-17:
Been reading through the examples and manual etc.  

I can't figure out how to sweep a 2 profile along a 3D curve.  Here's the goal: a parametric version of a screw, and the sweep is to generate the threads.  So assuming there is a cylinder with and cone at its end, I'd like to have a spiral path go up the length of the cylinder and sweep a triangular profile along it to generate the threads.  I'm stuck because, for example, polylines only have 2D coordinates.

Any advice is much appreciated!

RubenRubens, 2021-01-17:
This is a copy paste from another issue.

```python
import cadquery as cq

r = 3 # Radius of the helix
p = 1.5 # Pitch of the helix
h = 4 # Height of the helix

# Helix
wire = cq.Wire.makeHelix(pitch=p, height=h, radius=r)
helix = cq.Workplane(obj=wire)

# Final result. A circle sweeped along a helix.
result = (
    cq.Workplane('XZ')
    .center(r, 0)
    .circle(0.1)
    .sweep(helix, isFrenet=True)
)
```

daemacles, 2021-01-18:
Thank you - worked perfectly.  

There aren't any examples using `cq.Wire.makeHelix()` in the documents.  What's the process to add one?

jmwright, 2021-01-18:
@daemacles  If you are going to add a basic example on how to use `makeHelix()` you could create a PR and add the example as Ex025_* here: https://github.com/CadQuery/cadquery/tree/master/examples

If your example is more complex (i.e. a bolt/screw generator), it's probably best to add it to the contrib repo: https://github.com/CadQuery/cadquery-contrib

daemacles, 2021-01-19:
PR added #594 added.

closed 2021-01-19 by daemacles