CadQuery/cadquery #590: setup.py installation:  No module

r4dr3fr4d, 2021-01-17:
Is there a way to either install (to working condition) cadquery via setup.py *or* a way to co-install OCP since it doesn't have its own setup.py?  
```
>>> import cadquery
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.9/site-packages/cadquery-2.1rc1-py3.9.egg/cadquery/__init__.py", line 2, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "/usr/lib/python3.9/site-packages/cadquery-2.1rc1-py3.9.egg/cadquery/occ_impl/geom.py", line 5, in <module>
    from OCP.gp import gp_Vec, gp_Ax1, gp_Ax3, gp_Pnt, gp_Dir, gp_Trsf, gp_GTrsf, gp, gp_XYZ
ModuleNotFoundError: No module named 'OCP'
```

adam-urbanczyk, 2021-01-17:
You need to install OCP by hand (i.e. using conda, or compile it yourself). `Setup.py` is mostly meant for development of CQ. If you want to just use it, then use conda.

adam-urbanczyk, 2021-01-17:
This is a duplicate of #153.

closed 2021-01-17 by adam-urbanczyk

r4dr3fr4d, 2021-01-17:
I'd be willing to use conda if it, you know, like, worked at all.