CadQuery/cadquery #593: exportSvg doesn't hide edges that are partially visible

MarcSallent, 2021-01-18:
This code generates two boxes, one of them crossing the other one. When exporting to SVG with `exportSvg`, edges that start in a visible area and then _intersect_ the other object are rendered incorrectly.
If they _go behind_ the other object, they are rendered correctly.

**Incorrect**
    from cadquery import cq, Compound
    nl = list()
    nl.append(cq.Workplane("XY").box(.1, 2, .1, centered=(True, True, True)).val())
    nl.append(cq.Workplane("XY").box(2, 1, 1, centered=(True, True, True)).val())
    comp = Compound.makeCompound(nl)
    cq.Workplane(comp).exportSvg("C:/Users/Marc/Documents/beehut/cadquery_tests/export/boxes.svg")
    
[image]

**Correct**
    from cadquery import cq, Compound
    nl = list()
    nl.append(cq.Workplane("XY").box(.1, 2, .1, centered=(True , True, True)).**translate((0,0,-1)).val()**)
    nl.append(cq.Workplane("XY").box(2, 1, 1, centered=(True, True, True)).val())
    comp = Compound.makeCompound(nl)
    cq.Workplane(comp).exportSvg("C:/Users/Marc/Documents/beehut/cadquery_tests/export/boxes.svg")
    
[image]

Cadquery 2.0 - Python 3.7 - Windows,

MarcSallent, 2021-01-18:
To add more info doing Union on the solids works as expected

    ra = (cq.Workplane("XY")
          .box(.1, 2, .1, centered=(True , True, True))
          .union(cq.Workplane("XY")
                 .box(2, 1, 1, centered=(True, True, True))
                 )
          )
    ra.exportSvg("boxes3.svg")

adam-urbanczyk, 2021-01-18:
This seems to do the trick for me:
```python
from cadquery import cq, Compound
nl = list()
nl.append(cq.Workplane("XY").box(.1, 2, .1, centered=(True, True, True)).val())
nl.append(cq.Workplane("XY").box(2, 1, 1, centered=(True, True, True)).val())
comp = Compound.makeCompound(nl).fuse()

cq.Workplane(comp).exportSvg("boxes.svg")
```

adam-urbanczyk, 2021-01-18:
You want to have internal edges, I assume. Another bool op needs to be used. I think one from here: https://github.com/CadQuery/cadquery/issues/449

MarcSallent, 2021-01-18:
I will try #449 , but I now understand that an edge _crossing a solid through one of their faces_ (not holes) is not a natural occurence, so maybe this is not actually an issue / worth solving. 
Seeing that `fuse()` or `union()` perform as expected, I would understand that this issue is closed. (I was avoiding using them to make the export faster)

adam-urbanczyk, 2021-01-19:
I'll leave the issue open. It might be good to at leas mention this in the docs.

roipoussiere, 2021-04-16:
I have a similar problem, this code generates glitched svg images.

```py
import cadquery as cq
from cadquery.cq import Workplane as WP

def rod(axis):
    wp = WP('XY' if axis == 'x' else 'XZ')
    return wp.circle(1).extrude(8)

fig = WP('XY').box(5, 5, 5).add(rod('x')).add(rod('y'))

show_object(fig)
cq.exporters.export(fig, 'figure.svg', opt={
    'projectionDir': (0.3, -0.2, 0.1),
    'showHidden': False
})
```

CQ-editor VS svg-export:

[image][image]

marcus7070, 2021-04-16:
> Does it really make sense to have different behaviors for add and union?

I can see why a user would think they are related methods, but internally they are pretty much unrelated. If you look at the source you can see they do completely different things. It would be very rare you use add.

adam-urbanczyk, 2021-04-16:
The solution is to implement #449 and use it before SVG export. The HLR algo does not know what to do with interfering but unrelated solids. It is arguably a very weird use case.