CadQuery/cadquery #595: Apparent hang when cutting a helix from a shape

daemacles, 2021-01-19:
There seems to be a hang when cutting certain sized helixes from other objects.  Am I doing something quite wrong?

Please try the code below.

```
import cadquery as cq

r = 0.42 # Radius of the helix
p = 0.4 # Pitch of the helix

#### CHANGE TO 2.5 and no cut
#### CHANGE TO 3.0 and it hangs
h = 2.49 # Height of the helix

s=6 # scale factor
curve_pts = ((
    (0,0),
    (0.1, s*0.01), 
    (0.2, s*0.04), 
    (0.3, s*0.09), 
    (0.4, s*0.16),
    (0.4, 1.0), 
    (0.4, 1.5)
    ))

# Helix
wire = cq.Wire.makeHelix(pitch=p, height=h, radius=r)
helix = cq.Workplane(obj=wire)

shaft = (
    cq.Workplane('XY')
    .circle(0.1)
    .workplane(offset=0.5)
    .circle(0.4)
    .loft()
    .faces('>Z')
    .circle(0.4)
    .extrude(2.0)
    )
#show_object(shaft)

# Final result. A circle sweeped along a helix.
threads = (
    cq.Workplane('XZ')
    .center(r, -0.5)
    .polyline(((-.15,0.05), (0.0,0.01), (0,0.39), (-0.15,0.35)))
    .close()
    .sweep(helix, isFrenet=True)
)
#show_object(threads)

result = shaft.cut(threads)
show_object(result)
```

marcus7070, 2021-01-19:
https://github.com/CadQuery/cadquery/issues/407 might help.

daemacles, 2021-01-19:
Thanks - I took a look at #407 and by just adding `clean=False` in the `shaft.cut(threads)` made it work.  So somehow the clean operation is failing.

adam-urbanczyk, 2021-02-09:
This works for me even with clean:
```python
import cadquery as cq

r = 0.42 # Radius of the helix
p = 0.4 # Pitch of the helix

#### CHANGE TO 2.5 and no cut
#### CHANGE TO 3.0 and it hangs
h = 2.49 # Height of the helix

# Helix
wire = cq.Wire.makeHelix(pitch=p, height=h, radius=r)
helix = cq.Workplane(obj=wire)

shaft = (
    cq.Workplane('XY')
    .circle(0.1)
    .extrude(0.5, taper=-30)
    .faces('>Z')
    .wires().toPending()
    .extrude(2.0)
    )
#show_object(shaft)

# Final result. A circle sweeped along a helix.
threads = (
    cq.Workplane('XZ')
    .center(r, -0.5)
    .polyline(((-.15,0.05), (0.0,0.01), (0,0.39), (-0.15,0.35)))
    .close()
    .sweep(helix, isFrenet=True)
)
#show_object(threads)

result = shaft.cut(threads)
show_object(result)
```

adam-urbanczyk, 2021-02-09:
This seems to work fine:
```python
import cadquery as cq

r = 0.42 # Radius of the helix
p = 0.4 # Pitch of the helix

#### CHANGE TO 2.5 and no cut
#### CHANGE TO 3.0 and it hangs
h = 3 # Height of the helix

# Helix
wire = cq.Wire.makeHelix(pitch=p, height=h, radius=r)
helix = cq.Workplane(obj=wire)

shaft = (
    cq.Workplane('XY')
    .circle(0.1)
    .extrude(0.5, taper=-30)
    .faces('>Z')
    .wires().toPending()
    .extrude(2.0)
    )
#show_object(shaft)

# Final result. A circle sweeped along a helix.
threads = (
    cq.Workplane('XZ')
    .center(r, -0.5)
    .polyline(((-.15,0.05), (0.0,0.01), (0,0.39), (-0.15,0.35)))
    .close()
    .sweep(helix, isFrenet=True)
    .transformed((-90,0,0))
    .workplane(2.5)
    .split(False,True)
)
#show_object(threads)

result = shaft.cut(threads)
```