CadQuery/cadquery #600: travis-ci.org is Shutting Down

jmwright, 2021-01-22:
Saw the following when I was on travis-ci.org
[image]

Blog Post: https://mailchi.mp/3d439eeb1098/travis-ciorg-is-moving-to-travis-cicom

Here's an issue from another project that's having to decide what to do. They've noticed the same extremely long build times that I've noticed as well over the last few months: https://github.com/letscontrolit/ESPEasy/issues/3462

I guess there are two ways to approach this:
1. Simply migrate our travis-ci.org builds to travis-ci.com
2. Use this as an opportunity to consolidate our Travis CI/CD onto a platform like Azure, which we already use for some other things.

I feel like over time our CI/CD setup has become fragmented, but maybe there's a good reason for that. If there's not, maybe we should discuss option 2.

adam-urbanczyk, 2021-01-24:
Travis and Appveyor allow to log onto the build agent. That is not possible on Azure AFAIK.

jmwright, 2021-02-05:
Migration guide: https://docs.travis-ci.com/user/migrate/open-source-repository-migration/

techdragon, 2022-02-08:
Are there still any outstanding issues related to this?

jmwright, 2022-02-08:
Not to my knowledge, but if anyone sees any Travis references or config files anywhere, please post here.

closed 2022-02-08 by jmwright