CadQuery/cadquery #601: Examples 009 and 010 doesn't work

RubenRubens, 2021-01-24:
`.isValid()` return false.

[Example 009 error output](https://github.com/RubenRubens/cq-testing/runs/1757169214?check_suite_focus=true#step:4:165)
[Example 010 error output](https://github.com/RubenRubens/cq-testing/runs/1757169214?check_suite_focus=true#step:4:197)

adam-urbanczyk, 2021-01-24:
Use the docs examples: https://github.com/CadQuery/cadquery/blob/master/doc/examples.rst

@jmwright I think the examples folder is out of sync with the docs and we should have one source of truth - maybe we should just remove them?

jmwright, 2021-01-25:
I think those examples are in a directory mainly because it allowed us to load them in the FreeCAD workbench for ease of access to users. If we're fairly sure that CQ-editor or another GUI will not want to include them, we can eliminate those examples. However, there's at least one example that will need to be merged into the docs before we get rid of those. Ex025 was added very recently. We should also probably just delete the example files and put a readme in that directory with a link, telling users that the examples are included only in the docs now.

adam-urbanczyk, 2021-01-30:
There seem to be more examples not appearing in the docs. For now I fixed the offending and added tests.

PR #609 (2021-01-31): Fix examples and start testing them
Will resolve #601
closed 2021-01-31 by adam-urbanczyk