CadQuery/cadquery #605: cannot draw 2D square

brianng0305, 2021-01-29:
I cannot drawing double square in 2d format
`layout = cq.Workplane().rect(100,100).rect(90,90)`
[image]

But it works for extrude
`layout = cq.Workplane().rect(100,100).rect(90,90).extrude(10)`
[image]

marcus7070, 2021-01-29:
This is part of the internals of CadQuery.

CQ-editor shows what is currently in a Workplane's objects attribute.

After `layout = cq.Workplane().rect(100,100)`, the `layout.objects` list contains one square, and that square is also appended to the `layout.ctx.pendingWires` list.

After the second square, ie. `layout = cq.Workplane().rect(100,100).rect(90,90)`, the `layout.objects` list contains just one 90x90 square, and that square is also appended to the `layout.ctx.pendingWires` list (which now contains both squares). Extrude uses the `layout.ctx.pendingWires` list, while CQ-editor just displays the current `layout.objects`.

adam-urbanczyk, 2021-01-30:
If you _really_ need to see both at once you can consolidate them.
```
res = cq.Workplane().rect(1,1).rect(3,3).consolidateWires()
```

closed 2021-01-30 by adam-urbanczyk