CadQuery/cadquery #610: wedge centering

marcus7070, 2021-01-31:
I think the centering in `Workplane.wedge` is incorrect. I'd like to fix it but just want to make sure I'm not breaking anything, especially given I've never used `wedge` before.

```python
w = cq.Workplane().wedge(10, 10, 10, 4, 4, 6, 6, centered=(True, True, True))
```

[image]

Note that the bottom left corner is on the origin, which is what I would expect from `centered=(False, False, False)`.

Change centered to `(False, True, True)` and you get:

[image]

Now the shape doesn't touch the origin at all. The Y and Z directions also behave the same.

Shall I change it such that `centered=(True, True, True)` puts the center of a bounding box around the wedge on the origin?

FYI the last time the `wedge` method was significantly changed was in #228.

jmwright, 2021-01-31:
My vote would be to fix it so that it works correctly. I don't think wedge gets used very much and that's probably why it hasn't been noticed sooner. Looking at the code, I think it's just an oversight that causes it to not be centered correctly. In #228 I was focused on fixing other things and apparently overlooked the centering problem.

https://github.com/CadQuery/cadquery/blob/3348c18889b37b7489468b4a0f59d73494a67389/cadquery/cq.py#L3532

adam-urbanczyk, 2021-01-31:
> My vote would be to fix it so that it works correctly.

I agree, looks like an easy fix.

adam-urbanczyk, 2021-01-31:
Let's fix this and finally release 2.1

PR #611 (2021-02-01): Fix wedge centering
Will resolve #610
closed 2021-02-01 by jmwright