CadQuery/cadquery #618: forConstruction option in offset2d

marcus7070, 2021-02-05:
I often want to use `Workplane.offset2D` to create construction geometry. For example:

```python
board_dims = (100, 50)
w = (
    cq.Workplane()
    .box(*board_dims, 1)
    .faces(">Z")
    .workplane()
    .rect(*board_dims)
    .offset2D(-2.4)
    .vertices()
    .hole(3)
)
```

Which actually works just fine:
[image]
Until you go to use `w.ctx.pendingWires`, for example:
```python
board_dims = (100, 50)
w = (
    cq.Workplane()
    .box(*board_dims, 1)
    .faces(">Z")
    .workplane()
    .rect(*board_dims)
    .offset2D(-2.4)
    .vertices()
    .hole(3)
    .faces(">Z")
    .workplane()
    .circle(10)
    .extrude(1)
)
```
[image]
And now you realise `offset2D` leaves the shape in `pendingWires`.

Is it worth adding a `forConstruction` option, similar to `rect` and other methods?

adam-urbanczyk, 2021-02-05:
+1, good idea.

marcus7070, 2021-02-05:
I would think a typical use case would be:
```python
w = (
    cq.Workplane
    .rect(10, 10, centered=False, forConstruction=True)
    .offset2D(-1, forConstruction=True)
    .vertices()
    .hole(1)
)
```

So currently `offset2D` gets wires from `pendingEdges` and `pendingWires`.

https://github.com/CadQuery/cadquery/blob/184a985066284c3895dc58499c0b6bcafaaaa15f/cadquery/cq.py#L3675-L3693

To enable it to work off construction geometry, I think when `forConstruction=True`, `offset2D` should take wires from `self.objects` and not do anything with `pendingEdges` and `pendingWires`. Does that sound right @adam-urbanczyk? I want to make sure I get this right because I think this is the first method that transforms construction geometry, and I would hate to set a bad precedent.

jmwright, 2021-02-14:
@marcus7070 Are you blocked on this? You probably won't find a 1-to-1 comparison with offset2d in the codebase, but you could take a look at how `forConstruction` is handled currently other places to see if you can find a precedent.

adam-urbanczyk, 2021-02-14:
@marcus7070 I'm not sure what the possible issue is. Does something go wrong, if you just set the `forConstruction` field of the generated wire?

marcus7070, 2021-02-14:
Not blocked, thanks.

Out of the two methods (taking edges from `self.objects` or pending wires/edges when `forConstruction=True`), I decided that `self.objects` made more sense to me and wrote the code, tests (https://github.com/marcus7070/cadquery/commit/5e3fb9e039089d55c53ce4757129aaebecb5f288) and an example based off that (https://github.com/marcus7070/cadquery/commit/e3cdb52e26bfd8d7649f3149289566fac8fcd293).

Then I noticed that `Workplane.wire` already uses pending edges/wires. So I will rewrite my stuff to keep in line with the existing code.

marcus7070, 2021-02-14:
> I'm not sure what the possible issue is.

I had some misconceptions about pending wires and construction geometry. I was thinking that it should be possible to chain construction geometry calls together, like `w = Workplane().rect(1, 1, forConstruction=True).offset2D(1, forConstruction=True)`. I was also thinking that turning pending wires into construction geometry should not be allowed, since pending wires are:

https://github.com/CadQuery/cadquery/blob/d1ebfbac2278e1cbf626009fbf0568e38cba2279/cadquery/cq.py#L85-L87

But I've come around now, there is no other way to do the simple `Workplane().vLine(1).hLine(1).close().offset2D(1, forConstruction=True)` except to use pending edges.

I'll get the PR done today or tomorrow.

PR #639 (2021-02-19): offset2D for construction
Will close #618.

I added a new example that covers offset2D as well.
closed 2021-02-19 by marcus7070