CadQuery/cadquery #620: 'OCP' is not a package

bhowiebkr, 2021-02-07:
Sorry if this is asked before but I'm trying to get cadquery up and running and have tried both under anaconda + miniconda, using both master and 2.0 release in the installation docs. What am I missing to get this running? I'm trying to use this on Windows10. 

`(cadquery) C:\Users\howard\work\FreeCAD_python>C:/Users/howard/.conda/envs/cadquery/python.exe c:/Users/howard/work/FreeCAD_python/block.py        
Traceback (most recent call last):
  File "c:/Users/howard/work/FreeCAD_python/block.py", line 4, in <module>
    import cadquery as cq
  File "C:\Users\howard\.conda\envs\cadquery\lib\site-packages\cadquery\__init__.py", line 14, in <module>
    from .occ_impl import exporters
  File "C:\Users\howard\.conda\envs\cadquery\lib\site-packages\cadquery\occ_impl\exporters\__init__.py", line 8, in <module>
    from OCP.VrmlAPI import VrmlAPI
ModuleNotFoundError: No module named 'OCP.VrmlAPI'; 'OCP' is not a package`

jmwright, 2021-02-08:
Looks like this may be related to this issue on the CQ-editor repo. https://github.com/CadQuery/CQ-editor/issues/222

bhowiebkr, 2021-02-08:
Just tested with Ubuntu 20.04 and got the same error.

bhowiebkr, 2021-02-08:
for now I've just commented it out as a temp solution:
https://github.com/CadQuery/cadquery/blob/184a985066284c3895dc58499c0b6bcafaaaa15f/cadquery/occ_impl/exporters/__init__.py#L8
https://github.com/CadQuery/cadquery/blob/184a985066284c3895dc58499c0b6bcafaaaa15f/cadquery/occ_impl/exporters/__init__.py#L108

adam-urbanczyk, 2021-02-08:
I messed up releasing OCP 7.4.0, should be fixed today (package is already building, ETA 3h). Once done, you'll have to reinstall using this magic incantation (assuming you have cq-editor):
```
conda remove caquery cq-editor
conda clean --all --force-pkgs-dirs
conda install --force-reinstall -c conda-forge -c cadquery cq-editor=master
```

jmwright, 2021-02-08:
@bhowiebkr It has been reported that this has been fixed. Please try again.

closed 2021-02-10 by bhowiebkr