CadQuery/cadquery #623: What would be the best way to do such?

MarcWeber, 2021-02-09:
What I am looking for is creating kind of bell for a trumpet but should have irregular shape.
It's like using 2 polylines within a plane to define outer/inner shape and then extruding it but also giving the next level a different polyline.
It's like morphing a star into a circle like loft but with more levels and changing diameter and extrude along a line.
if this is complicated is it better to create .stl like with Python code combining vertexes to faces and importing that or can cadquery turn points into faces as well ?

adam-urbanczyk, 2021-02-09:
I think this is what you want:
https://github.com/CadQuery/cadquery/blob/184a985066284c3895dc58499c0b6bcafaaaa15f/tests/test_cadquery.py#L871

CQ has no surface fitting functions implemented.

RubenRubens, 2021-02-09:
This might help as well:
https://github.com/RubenRubens/cq-testing/blob/master/passing/loft/L.py
[image]

It is a loft from a circle and two ellipses.

bragostin, 2021-02-10:
How about this?
```python
import cadquery as cq
import math

r1 = 1.0
thickness = 0.1
fn = 6
edge_points = [
    [
        r1 * math.cos(i * 2 * math.pi / fn + 30 * math.pi / 180),
        r1 * math.sin(i * 2 * math.pi / fn + 30 * math.pi / 180),
    ]
    for i in range(fn + 1)
]
surface_points = [
    [
        r1 / 4 * math.cos(i * 2 * math.pi / fn + 30 * math.pi / 180),
        r1 / 4 * math.sin(i * 2 * math.pi / fn + 30 * math.pi / 180),
        0.75,
    ]
    for i in range(fn + 1)
] + [[0, 0, 2]]
edge_wire = cq.Workplane("XY").polyline(edge_points)
plate_3 = (
    cq.Workplane("XY")
    .interpPlate(
        edge_wire,
        surface_points,
        thickness,
        combine=False,
        clean=False,
        degree=2,
        nbPtsOnCur=20,
        nbIter=2,
        anisotropy=False,
        tol2d=0.00001,
        tol3d=0.0001,
        tolAng=0.01,
        tolCurv=0.1,
        maxDeg=8,
        maxSegments=9,
    )
)
```
[image]

adam-urbanczyk, 2021-02-10:
Yet another option - adding it to show shelling. As you can see there are many ways to skin a cat in CQ ;).
```python
res = (
   cq.Workplane()
   .circle(2)
   .workplane(offset=10)
   .polygon(8,20)
   .loft()
   .faces('>Z or <Z').shell(.5)
)
```
[image]

dcowden, 2021-02-10:
Great example Adam! Always amazing how little code it can take to make an
object!
This  might be a great one for the examples!

MarcWeber, 2021-07-07:
How would you implement such extrusion along path ?

Naiv approach like this looks funny at best.
Without rotateAboutCenter it looks fine.

```
r = cq.Workplane("XZ").circle(2).workplane(offset=2).circle(1).workplane(offset=2).circle(2).rotateAboutCenter([0,0,1], 10).loft()
```