CadQuery/cadquery #624: dxf layer import exclusion not matching layer names properly

greyltc, 2021-02-09:
I'm having a problem with https://github.com/CadQuery/cadquery/pull/442 in that ezdxf is returning layer names that are all upper case, so none of my layer name exclusions are matching.

Not really sure yet why this happening now, maybe exdxf changed or the DXF file format I'm using is somehow causing the layer names to become upper case.

Need to investigate a bit more. Maybe we should just force all the layer names we deal with to lower case to avoid case issues.

greyltc, 2021-02-09:
Definitely depends on what flavor of .dxf you feed in (there are a lot of flavors, so that's annoying). I wonder if this could/should be solved in ezdxf

jmwright, 2021-02-09:
@greyltc If you think ezdxf may have changed this functionality in newer versions, you could try pinning an older version [here](https://github.com/CadQuery/cadquery/blob/master/environment.yml#L18) and rebuild the Anaconda environment. DXF import is relatively new so it may just be that it's always been that way and nobody has noticed it until now.

Enforcing a case for layer names sounds reasonable, but there may be problems with the approach that I'm not considering.

adam-urbanczyk, 2021-02-09:
Can you check on the ezdxf bug tracker? Seems that they do some "normalization":
https://github.com/mozman/ezdxf/blob/ea6775e143454bb8cb5cbe19f42d5f351c1bd326/src/ezdxf/addons/drawing/properties.py#L247

adam-urbanczyk, 2021-02-10:
@greyltc I tried to reproduce the issue locally, but cannot. I have a DXF with upper and lower case names and filtering on it works as expected. Can you share your file?

greyltc, 2021-02-10:
Yeah, I'll try to make up a minimally reproducible example for you, with file.
I believe this was with a .dxf file I drew and saved with my drawing software as type `R14 ASCII drawing (*.dxf)`

greyltc, 2021-02-10:
Ok, I've made a minimal example files with a circle on `laYer_zero`, and a vertical line on `laYer_one` and saved that as file name `R14_ASCII_layer_test.dxf` of type "R14 ASCII drawing (\*.dxf )"
Then I saved it again as file name `R2018_ASCII_layer_test.dxf` as type "R2018 ASCII Drawing (\*.dxf)"

After inspecting the contents of the files, it turns out that the layer name strings are actually all uppercase in the R14 .dxf file format (they're as expected in the R2018 one). So the case information is actually discarded at save time. Maybe the R14 ASCII standard doesn't support case sensitive layer names?
[layer_name_test.zip](https://github.com/CadQuery/cadquery/files/5960118/layer_name_test.zip)

mbway, 2021-02-10:
as far as I'm aware, dxf layer names are not case sensitive in any version so they are normalised (to lower case) for use in the rendering context, but ezdxf returns the layer name in it's original case when you query it

greyltc, 2021-02-11:
> as far as I'm aware, dxf layer names are not case sensitive in any version
Then it seems we should force case agreement here while we're comparing strings.

adam-urbanczyk, 2021-02-11:
@greyltc let's not be too hasty on this topic. It seems that there **is** software that does preserve names and ezdxf does seem to support it. I also cannot find anything in the spec about names (it is written in a terrible way though). From this point of view I could say that the root cause is related to the tool you used to save your DXF and not to CQ. Would you agree?

mbway, 2021-02-11:
if you want I can see what AutoCAD does with the dxf file where the layers differ only in case. From what I can remember it will treat the layers as the same. There may be other CAD applications which are case sensitive but these would be non-compliant and so not worth considering IMO.

ezdxf lets you read the values exactly as they are stored in the dxf file but it's up to the application to handle the data properly, the same way that a html parser might let you see tag names with their original case but the browser is still case insensitive.

adam-urbanczyk, 2021-02-11:
@mbway that'd be great.
@greyltc  which software did you use to generate the two files?

greyltc, 2021-02-12:
I generated them with DraftSight.

I've since tested AutoCAD and it behaves the same way (layer name strings converted to all upper case at save time, but only when saving as certain DXF flavors).

I can also say that neither software (DraftSight nor AutoCAD) allows the user to name/create layers that differ only in case.

mbway, 2021-02-12:
more evidence is that this crashes:
```python
import ezdxf

def main():
doc = ezdxf.new()
layout = doc.modelspace()

doc.layers.new('ABc')
doc.layers.new('abc')  # ezdxf.lldxf.const.DXFTableEntryError: LAYER abc already exists!
```
and when this file is opened in autocad the entities appear to be on the same layer:
```python
import ezdxf

doc = ezdxf.new()
layout = doc.modelspace()

doc.layers.new('ABc')

layout.add_line((0, 0), (0, 1), {'layer': 'AbC'})
layout.add_line((1, 0), (1, 1), {'layer': 'abc'})

doc.saveas('/tmp/test.dxf')
```
[test.dxf.zip](https://github.com/CadQuery/cadquery/files/5971088/test.dxf.zip)

[image]

adam-urbanczyk, 2021-02-12:
OK, thanks for the thorough investigations. I'm convinced!

PR #630 (2021-02-13): force dxf layer name case agreement
fixes #624
commit 42733314 (2021-02-13): force dxf layer name case agreement

fixes #624
closed 2021-02-13 by jmwright