CadQuery/cadquery #628: Array an object but with varrying parameters

jalovisko, 2021-02-10:
Hello. I am exploring CadQuerry and parametrization is really well customized. I have a question though, is it possible to do something similar to what has been covered in issue #586 but with varrying parameter? Lets say I want the Ls from that issue to have different diameter each.

marcus7070, 2021-02-10:
Sure. Here is a quick and __dirty__ solution:

```python
import cadquery as cq
from random import random

def createLs(self, diam=10):
    L = cq.Workplane().circle(diam / 2).extrude(-500)
    L = L.transformed(offset=cq.Vector(0,0,-500)).transformed(rotate=cq.Vector(90,0,0)).circle(diam / 2).extrude(-500)

    return self.union(self.eachpoint(lambda loc: L.val().located(loc), True))

cq.Workplane.createLs = createLs

pnts = [(0, 0), (100, 0), (200, 0)]
diams = [5 + 50 * random() for _ in range(len(pnts))]

L2 = cq.Workplane().tag('base')

for pnt, diam in zip(pnts, diams):
    L2 = L2.workplaneFromTagged('base').center(*pnt).createLs(diam)

show_object(L2)
```
[image]

I should say this solution is all about the fact that CadQuery is a Python library, and you can leverage any technique from the whole world of Python to make building your models easier. These methods (monkey patching, passing arguments) are general Python techniques, not unique to CQ.

adam-urbanczyk, 2021-02-10:
I think the canonical way of doing that would be to use `cq.Wokrplane.eachpoint`:
```python
eachpoint(
    self,
    callback: Callable[[cadquery.occ_impl.geom.Location], cadquery.occ_impl.shapes.Shape],
    useLocalCoordinates: bool = False,
) -> 'Workplane'
```

Then it is up to you to:
1) put the desired locations on the stack using e.g. `pushPoints` or `rarray`;
2) provide a callable that accepts a `cq.Location` object and returns the desired `cq.Shape`. This callable can be of course stateful, or you can implement a logic that maps from a `cq.Location` to you object properties.

It has also cross my mind to enable `cq.Location` to store arbitrary metadata. Due to the dynamic nature of Python you could try it yourself.

jalovisko, 2021-02-16:
Amazing, thanks @marcus7070 and @adam-urbanczyk!

closed 2021-02-16 by jalovisko

tanius, 2021-03-25:
I just had the same challenge. I like the canonical `Workplane::eachpoint()` solution but wanted a way to pass in additional arguments to the callback. So I made an extended version called `eachpointAdaptive()`. Show to create the L's from #586, with varying diameters as the OP proposed:

```python
import cadquery as cq

def eachpointAdaptive(
    self,
    callback,
    callback_extra_args = None,
    useLocalCoords = False
):
    """
    Same as each(), except that (1) each item on the stack is converted into a point before it
    is passed into the callback function and (2) it allows to pass in additional arguments, one 
    set for each object to process.

    Conversion of stack items into points means: the resulting stack has a point for each object 
    on the original stack. Vertices and points remain a point. Faces, Wires, Solids, Edges, and 
    Shells are converted to a point by using their center of mass. If the stack has zero length, a 
    single point is returned, which is the center of the current workplane / coordinate system.

    :param callback_extra_args: Array of dicts for keyword arguments that will be 
        provided to the callback in addition to the obligatory location argument. The outer array 
        level is indexed by the objects on the stack to iterate over, in the order they appear in 
        the Workplane.objects attribute. The inner arrays are dicts of keyword arguments, each dict 
        for one call of the callback function each. If a single dict is provided, then this set of 
        keyword arguments is used for every call of the callback.
    :param useLocalCoords: Should points provided to the callback be in local or global coordinates.

    :return: CadQuery object which contains a list of vectors (points) on its stack.

    .. todo:: Implement that callback_extra_args can also be a single dict.
    .. todo:: Implement that empty dicts are used as arguments for calls to the callback if not 
        enough sets are provided for all objects on the stack.
    """

    # Convert the objects on the stack to a list of points.
    pnts = []
    plane = self.plane
    loc = self.plane.location
    if len(self.objects) == 0:
        # When nothing is on the stack, use the workplane origin point.
        pnts.append(cq.Location())
    else:
        for o in self.objects:
            if isinstance(o, (cq.Vector, cq.Shape)):
                pnts.append(loc.inverse * cq.Location(plane, o.Center()))
            else:
                pnts.append(o)

    # If no extra keyword arguments are provided to the callback, provide a list of empty dicts as 
    # structure for the **() deferencing to work below without issues.
    if callback_extra_args is None:
        callback_extra_args = [{} for p in pnts]

    # Call the callback for each point and collect the objects it generates with each call.
    res = []
    for i, p in enumerate(pnts):
        p = (p * loc) if useLocalCoords == False else p
        extra_args = callback_extra_args[i]
        p_res = callback(p, **extra_args)
        p_res = p_res.move(loc) if useLocalCoords == True else p_res
        res.append(p_res)

    # For result objects that are wires, make them pending if necessary.
    for r in res:
        if isinstance(r, cq.Wire) and not r.forConstruction:
            self._addPendingWire(r)

    return self.newObject(res)

def create_l(location, diameter):
    result = (
        cq.Workplane()

        # Vertical part.
        .circle(diameter / 2)
        .extrude(-500)

        # Horizontal part.
        .transformed(offset = cq.Vector(0,0,-500))
        .transformed(rotate = cq.Vector(90,0,0))
        .circle(diameter / 2)
        .extrude(-500)
    )

    return result.val().located(location)

# Register our custom plugin before use.
cq.Workplane.eachpointAdaptive = eachpointAdaptive

model = (
    cq.Workplane("XY")
    .pushPoints([(0, 0), (100, 0), (200, 0)])
    .eachpointAdaptive(
        create_l,
        callback_extra_args = [{"diameter": 20}, {"diameter": 40}, {"diameter": 80}],
        useLocalCoords = True
    )
)

show_object(model) # Assumes file is executed in cq-editor.
```
If this or a similar solution sounds generic and useful enough to include into CadQuery, I'm happy to work on a PR.

adam-urbanczyk, 2021-03-25:
Maybe something for https://github.com/CadQuery/cadquery-plugins?