CadQuery/cadquery #631: `Workplane.spline()` Doesn't Support Tangents at Arbitrary Interpolation Points

pavpen, 2021-02-12:
Hello, developers.

The implementation of [`Workplane.spline`](https://github.com/CadQuery/cadquery/blob/b6beba7cb1b3d7cbd4dc9f65733ff24c8cd4fc62/cadquery/cq.py#L1670), doesn't seem to support specifying tangents for arbitrary interpolation points, but only at the start and end points. The underlying [`GeomAPI_Interpolate` interpolator does support this feature](https://dev.opencascade.org/doc/refman/html/class_geom_a_p_i___interpolate.html#a9379c1ad3ae8da85f066216912570fb1). (It is also [exposed in FreeCAD](https://github.com/FreeCAD/FreeCAD/blob/ab4ab444b66f675a2c8be39b41ad318f10998d82/src/Mod/Part/App/BSplineCurvePyImp.cpp#L975).)

I would like to add this feature to `Workplane.spline()`, or, perhaps, to a new method, if changing `spline()` may cause undesirable breakages.

Let me know, if this sounds like a good idea.

Thanks,
___
Pav

adam-urbanczyk, 2021-02-12:
Hey, absolutely! If you want to implement it, I'd propose to just check the length of the tangents in `cq.Workplane.spline` and in `cq.Edge.makeSpline` and act accordingly.

pavpen, 2021-02-13:
That's what I was doing. I was also adding the `parameters` argument (I need to figure out how to add some unit tests.):

```Python
    @classmethod
    def makeSpline(
        cls: Type["Edge"],
        listOfVector: List[Vector],
        tangents: Optional[Sequence[Vector]] = None,
        periodic: bool = False,
        parameters: Optional[List[float]] = None,
        tol: float = 1e-6,
    ) -> "Edge":
        """
        Interpolate a spline through the provided points.
        :param cls:
        :param listOfVector: a list of Vectors that represent the points
        :param tangents: tuple of Vectors specifying start and finish tangent
        :param periodic: creation of peridic curves
        :param parameters: the value of the parameter at each interpolation point.
            (The intepolated curve is represented as a vector-valued function of a
            scalar parameter.)
                If periodic == True, then len(parameters) must be
            len(intepolation points) + 1, otherwise len(parameters) must be equal to
            len(interpolation points).
        :param tol: tolerance of the algorithm (consult OCC documentation)
                Used to check that the specified points are not too close to each
            other, and that tangent vectors are not too short. (In either case
            interpolation may fail.)
        :return: an Edge
        """
```

Also, I think passing `Scale = False` to `GeomAPI_Interpolate.Load()` would save some unnecessary computation. But that's a minor point.

marcus7070, 2021-02-18:
Closed in #636

closed 2021-02-18 by marcus7070