CadQuery/cadquery #632: Drawing a coil with a square shaped cross section

n2d7, 2021-02-12:
Hi,
Is there a way to draw a coil with a square cross section and given number of rotations? Appreciate any help on this. 

Thanks

jmwright, 2021-02-12:
There is a swept helix example [here](https://github.com/CadQuery/cadquery/blob/master/examples/Ex025_Swept_Helix.py). I think you'll need to work out the values to give the number of turns yourself.

It's modified below to have a rectangular cross-section:
```python
import cadquery as cq

r = 0.5  # Radius of the helix
p = 0.4  # Pitch of the helix - vertical distance between loops
h = 2.4  # Height of the helix - total height

# Helix
wire = cq.Wire.makeHelix(pitch=p, height=h, radius=r)
helix = cq.Workplane(obj=wire)

# Final result: A 2D shape swept along a helix.
result = (
    cq.Workplane("XZ")  # helix is moving up the Z axis
    .center(r, 0)  # offset isosceles trapezoid
    .rect(0.15, 0.15)
    .sweep(helix, isFrenet=True)  # Frenet keeps orientation as expected
)

show_object(result)
```

n2d7, 2021-02-13:
Thank you. This works perfectly.
Additional question, is there a way to vary the radius dynamically? So that I can make a cone shaped spiral.

n2d7, 2021-02-13:
Found it. 
`wire = cq.Wire.makeHelix(pitch=p, height=h, radius=r, angle=30)`

jmwright, 2021-02-13:
@n2d7 It sounds like this is resolved for you, but feel free to reopen if it's not.

closed 2021-02-13 by jmwright