CadQuery/cadquery #634: NameError: name 'show_object' is not defined

n2d7, 2021-02-13:
I installed both cadquery and cq-editor in my anaconda environment. But when I run an example script, it gives the error that NameError: name 'show_object' is not defined. 
Am I doing something wrong?

tanius, 2021-02-13:
It may be that `show_object()` is used in an included file (using Python's `import …` statement). That is not possible as far as I know; `show_object()` is made available by cq-editor only to the file opened in the editor.

mbway, 2021-02-14:
something you can do is put this at the top of your script and `show_object()` will have no effect when running outside of CQ-editor:
```python
if 'show_object' not in globals():
    def show_object(*args, **kwargs):
        pass
```
if you want to use show_object inside non-main modules you can do this (it's a bit of a hack though)
```python
# (main module)
import my_other_module

my_other_module.show_object = show_object

...
```

closed 2021-06-26 by marcus7070

qkum, 2022-05-06:
> something you can do is put this at the top of your script and `show_object()` will have no effect when running outside of CQ-editor:
>
> ```python
> if 'show_object' not in globals():
>     def [...]

sorry for asking stupid.

what is a main and non-main modules in this context? (with examples pls)

mbway, 2022-05-07:
When referring to regular python scripts, the main module is the one that is passed to the interpreter to run. This module can then import other modules.

In the context of CQ-Editor, the editor typically operates on a single script (this is the main module) but it is possible to import other modules on the filesystem. However CQ-Editor only populates the `globals()` dictionary of the main module, so to use `show_object` in an imported module you have to manually inject it as I showed in the second snippet.

So an example might be:

```python
# main.py
import cadquery as cq
import cube_builder

if 'show_object' not in globals():
    def show_object(*args, **kwargs):
        pass

cube_builder.show_object = show_object

cube_builder.build_cube(1)
```

```python
# cube_builder.py
import cadquery as cq

def build_cube(size):
    cube = cq.Workplane('XY').box(size, size, size)
    show_object(obj)

```

This example is quite contrived but there are situations where you have re-usable functionality that you want to keep in a shared module and you may want to use `show_object` for debugging purposes.

The trick above may not be relevant for jupyter-cadquery. From [the README](https://github.com/bernhard-42/jupyter-cadquery) it seems like you should be importing `show_object` like so:
```python
from jupyter_cadquery.viewer.client import show_object
```

qkum, 2022-05-07:
> When referring to regular python scripts, the main module is the one that is passed to the interpreter to run. This module can then import other modules.
>
> In the context of CQ-Editor, the editor [...]

programming is so hard lol - (english as 2d language does not help)

i pretty much get what you are saying. 

i will try and figure out how i exactly execute it in real life, in my environment.

something with importing python file #1 into python file #2. and in python file #1 you imported show_object into a brand new class or object, that you then import into the python file #2 and thereby bypass/hack around the limitation of the non-logical static CQ-editor code limitations.

i think i got it 👍  
thanks for the fast and thorough response.

qkum, 2022-05-07:
PS: the reason i badly want to use visual studio code + jupyter notebook is the MUCH better syntax colors (read-able) and jupyter notebook is just a truly genius programming platform. so much time saved and less headache with this combo. (recommended big-time)

qkum, 2022-05-07:
the real solution/fix to me seems to be finding the folder in CQ_editor where the modules/libs/scripts/helper files are, and copy-paste them into another folder, and import them from there to anaconda/jupyter notebook.

but that is likely because im not advanced enough to do it the way you talk about.