CadQuery/cadquery #635: Help running on raspberry pi

rodrykthingol, 2021-02-13:
Hello, I'm a maker and i used to work with openscad. Then i found cadquery, loved it and used it as an excuse to learn python, as well as to model my crazy ideas.

The issue is that i do most of my fine tuning of the model on a raspberry pi. Is there any easy way to achieve this?

jmwright, 2021-02-13:
We don't have a way to install CadQuery on a Raspberry Pi since they run on ARM architecture. If you figure it out, please let us know.

If you're using an RPi as a desktop replacement and just want a way to tinker with models, you could try running jupyter-cadquery via MyBinder in Firefox on the Pi.

https://github.com/bernhard-42/jupyter-cadquery#quick-use-via-binder

I think you'll have to copy the text of your finished models manually into files on your local computer though. There's no way to save models on MyBinder.

adam-urbanczyk, 2021-02-15:
@rodrykthingol you probably need to compile a lot of things on your own. For sure OCCT, OCP and their deps (if not present in the distro).

adam-urbanczyk, 2023-03-17:
If there is still interest, open an issue in the OCCT/OCP conda-forge feedstock repos.

closed 2023-03-17 by adam-urbanczyk

twisted-owl, 2025-04-25:
Do you know if this was ever fixed? @adam-urbanczyk  I am trying to use a Raspberry PI for cadquery-OCP and I am running into issues.