CadQuery/cadquery #637: move does not work with wires

adam-urbanczyk, 2021-02-14:
E.g. `cq.Workplane().rect(10,10).move(10,10)`

fedorkotov, 2021-03-04:
You probably meant `cq.Workplane("XY").move(10,10).rect(10,10)`. 
[image]

[`move(..)`](https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=move#cadquery.Workplane.move) does not move previously created wires. It moves current point from which you can continue to draw new primitives. Don't know if it is a bug or not but you can not call move directly after `rect(..)`. But this works
```python
r = \
    cq.Workplane("XY")
      .rect(10,10)\
      .workplane()\
      .move(10,10)\
      .rect(10,10)\
      .extrude(10)
```

[image]

To move previously defined primitives you can use [`translate(..)`](https://cadquery.readthedocs.io/en/latest/classreference.html#cadquery.Workplane.translate)
This code `cq.Workplane("XY").rect(10,10).translate(cq.Vector(10,10,0))` produces the same result as `cq.Workplane("XY").move(10,10).rect(10,10)`

fedorkotov, 2021-03-04:
Sorry. Just noticed that you are one of the main contributors of the project. If you say `move()` does not work, than it most likely does not work indeed. I've just started using CadQuery a few days ago.
But still. What result did you expect to get from `cq.Workplane().rect(10,10).move(10,10)`?

dcowden, 2021-03-04:
I expect a rectangle, with the current working coordinates shifted by 10,10, such that the next object created would be centered at that newly shifted point. IE, try

```cq.Workplane().rect(10,10).move(50,50).circle(5)```

jmwright, 2021-03-04:
Trying the code @dcowden posted in CQ-editor 0.2 results in the following error.

```
RuntimeError: Cannot convert object type '' to vector.
```

I'm assuming the reason @adam-urbanczyk posted this issue is that the code above should work without erroring, and `move` should effect the next `circle` call as expected.

Here's my full code:

```
import cadquery as cq

result = cq.Workplane().rect(10,10).move(50,50).circle(5)

show_object(result)
```

dcowden, 2021-03-04:
yeah i mean, i _think_ that's what should happen-- but i've not been in the thick of it recently so feel free to correct me!

adam-urbanczyk, 2021-03-04:
Indeed, the issue is about the  `RuntimeError` and not semantics.