CadQuery/cadquery #643: Expose BRepOffset_MakeOffset for cq.Face

adam-urbanczyk, 2021-02-17:
This will allow to make thick solids from arbitrary faces.

gumyr, 2021-12-31:
Is this the type of thing being considered?  If so I could create a PR.
```python
def _thickenFace(self, thickness: float) -> cq.Solid:
    """Create a solid from a potentially non planar face by thickening along the normals"""
    solid = BRepOffset_MakeOffset()
    solid.Initialize(
        self.wrapped,
        thickness,
        1.0e-5,
        BRepOffset_Skin,
        False,
        False,
        GeomAbs_Intersection,
        True,
    )
    solid.MakeOffsetShape()
    return cq.Solid(solid.Shape())

cq.Face.thicken = _thickenFace

def _thickenSurface(self, thickness: float):
    """Find all of the faces on the stack and make them Solid objects by thickening along the normals"""
    return self.newObject([f.thicken(thickness) for f in self.faces().vals()])

cq.Workplane.thickenSurface = _thickenSurface
```

adam-urbanczyk, 2022-01-01:
Yes, that is is more or less it. I'll have some remarks on what is exposed and naming though.

gumyr, 2022-01-14:
After working with this for quite a while now, I've come to the conclusion that thicken needs help in determination of the polarity of the `depth` parameter - as follows:
```python
def _faceThicken(self, depth: float, direction: cq.Vector = None) -> cq.Solid:
    """
    Create a solid from a potentially non planar face by thickening along the normals.
    The direction vector can be used to potentially flip the face normal direction such that
    many faces with different normals all go in the same direction (direction need only be
    +/- 90 degrees from the face normal.)
    """

    # Check to see if the normal needs to be flipped
    adjusted_depth = depth
    if direction is not None:
        face_center = self.Center()
        face_normal = self.normalAt(face_center).normalized()
        if face_normal.dot(direction.normalized()) < 0:
            adjusted_depth = -depth

    solid = BRepOffset_MakeOffset()
    solid.Initialize(
        self.wrapped,
        Offset=adjusted_depth,
        Tol=1.0e-5,
        Mode=BRepOffset_Skin,
        # BRepOffset_RectoVerso - which describes the offset of a given surface shell along both
        # sides of the surface but doesn't seem to work
        Intersection=True,
        SelfInter=False,
        Join=GeomAbs_Intersection,  # Could be GeomAbs_Arc,GeomAbs_Tangent,GeomAbs_Intersection
        Thickening=True,
        RemoveIntEdges=True,
    )
    solid.MakeOffsetShape()
    try:
        result = cq.Solid(solid.Shape())
    except StdFail_NotDone as e:
        raise ValueError("Error applying thicken to given Face") from e

    return result

cq.Face.thicken = _faceThicken
```
Without this help, the face normal is the only guide and I've found that the normals are not consistent when creating non-planar faces where some characters having flipped normals from the other.  This is extremely frustrating as a user.

Here is an example where I emboss (not project) a long string of text on a cylinder:
[image]
The `thicken` code is called as follows:
```python
    projected_text = cq.Compound.makeCompound(
        [f.thicken(5, f.Center()) for f in projected_text_faces]
    )
```
and here is an example of projecting when the direction is easy to define:
```python
    projected_text = cq.Compound.makeCompound(
        [f.thicken(5, direction=cq.Vector(0, -1, 0)) for f in projected_text_faces]
    )
```
The try/except block is there to help avoid the dreaded "StdFail_NotDone" exceptions which can be frustrating for users. At least this indicates exactly where the failure happened.

Any feedback before I create a PR?

Thanks,
Roger

gumyr, 2022-01-14:
In addition - let me show you results that aren't as good:
[image]
The Face (with I confirmed to be valid) has a complex internal hole. The thickness varies widely across the surface from what looks to be near zero to about double the target thickness. I've tried all the `BRepOffset_MakeOffset` options that I could be get rid of these artifacts with no success.

I think this feature has great value even with these issues, but I just wanted you to see the bad as well as the good.

gumyr, 2022-01-14:
1. `direction` is there because the user can't know if the flipping needs to happen or not (other than forcing them to do the adjusted_depth calculation themselves - which is asking too much). `depth` can be both positive and negative (should there be a check against zero?) which statically controls the thickening direction which is what a bool would do.
2. The exception is not "swallowed", the new exception is raised from the `StdFail_NotDone` generic exception.
3. I'm not convinced that "thicken" is the best method name either.  I looked for synonym of extrude and got: force out, thrust out, squeeze out, express, press out, eject, expel, release, give off, emit, void, exude, excrete - none of which seem any better. "offset" implies to me a second Face parallel to the given Face which is actually what `BRepOffset_MakeOffset` does if the `Thickening` parameter isn't True.

gumyr, 2022-01-14:
Oh - are you concerned about non-even thickness in the flag example?

adam-urbanczyk, 2022-01-14:
No offense, but I really don't want to discuss 1+2.

gumyr, 2022-01-14:
Sorry, but changing direction to a bool just doesn't work. Try out the code and you'll see for yourself - I only introduced direction after working with the code in the real world.

closed 2022-07-08 by adam-urbanczyk