CadQuery/cadquery #645: SVG export for assembly objects

adam-urbanczyk, 2021-02-18:
Nice to have for generating docs and presentations.

roipoussiere, 2021-04-16:
I totally support this!

For my use-case, I want svg files with several parts in different colors. For now I export unify these parts in a single solid then I export it to svg, and manually color the parts in Inkscape, which is tedious.

The possibility to directly export a svg file with the colors defined in the assembly (ie. `cq.Assembly().add(fig1, color=cq.Color('red'))`) will be very convenient for me.

lawrencesayercode, 2022-03-24:
Great feature idea. 

The use case I had in mind when starting to use cadquery was to generate nice 2D dimensioned drawings from an assembly. I was excited to find the SVG export feature and was really pleased with the results it was giving for single parts. 

If it were possible, as it is with parts, to export an .svg, with a bit of extra python/inkscape messing-about, to prepare such drawings would have been feasible. Alas though, for assemblies it seems not yet possible!

jmwright, 2022-03-25:
I know this doesn't help with the multi-color aspect, but I'll leave this comment here for anyone who just wants to do a simple assembly export to SVG.

To generate an SVG for an entire assembly, it is be possible to call `toCompound()` on the assembly and then export it.
```python
import cadquery as cq

assy = cq.Assembly()
box1 = cq.Workplane().box(10, 10, 10)
box2 = cq.Workplane().center(10, 10).box(10, 10, 10)
assy.add(box1)
assy.add(box2)

cq.exporters.export(assy.toCompound(), '/home/jwright/Downloads/assy.svg')
```
You can then use the [SVG export options](https://cadquery.readthedocs.io/en/latest/importexport.html#exporting-svg) provided in the docs to get different views of the assembly.

roipoussiere, 2022-10-04:
It could be very useful to also have the assembly part names (ie. `.add(box1, name='box1')`) in the svg file as group ids (ie. `<g id="box1"`).
This will allow to add some post-processing in the Python script after the export.

nomad-vagabond, 2023-11-09:
Adding a fill option to svg export will be also useful