CadQuery/cadquery #646: Length selector

marcus7070, 2021-02-19:
I think it's worth making a length selector (Nth selector style, since that also covers min/max with `[0]`/`[-1]`).

The [Paramak project have their own version](https://github.com/ukaea/paramak/blob/785c3ed7304e22eac7d58bb1bdc6515fbd20b9a8/paramak/utils.py#L696-L736) and so does [cq-kit](https://github.com/michaelgale/cq-kit/blob/fd3868593d9610446a4400ab64b08eddd895ba79/cqkit/cq_selectors.py#L254-L288).

A length selector also would help out in #638, where @mbway had to select the inner of two concentric b-splines. 

Thumbs up for yes, thumbs down if you think it's just bloat.

jmwright, 2021-02-19:
+1 from me

Tagging @michaelgale (cq-kit) and @Shimwell (Paramak) so they can add their thoughts if they want.

Shimwell, 2021-02-19:
Yes please that sounds most useful. I have the situation where I have to find edges and I don't know much about it. Having a selector for the Nth largest and Nth smallest would be most useful.

I tend to search for these edges and then perform operations like fillet and currently the Paramak has a selector for a given length +- a tolerance which we would be delighted to move to CQ if it is better placed there.

The only complication I can think of is that sometimes the edges are the same length. So perhaps it would be interesting to have another filter that returns edges that are the same length (within a tolerance). Otherwise it might be a bit unpredictable as to which of the similar length edges are returned.

I think that the creation of selectors in these two code bases that rely on CQ is an indication that such tools are universally useful. I would happily use a selector from CQ and remove the one in the Paramak.

Keen to help with this effort if I can be of assistance.

ps great work @michaelgale :light_rail: :train: :train: :train: :train:

michaelgale, 2021-02-19:
Yes, a LengthSelector is indeed a useful tool and often used and therefore a helpful addition to CQ.  My only concern in terms of CQ-Kit is the namespace collision of the `LengthSelector` classname.  Since CQ-Kit uses the same classname there could be issues depending on the order of imports etc.

I could rename my base classname `LengthSelector` to avoid conflict if that is the name the CQ project would like to adopt.  Either way, let me know so that I can make changes.  It looks like it will be a painless and non-breaking change for CQ-Kit.
p.s. thanks @Shimwell !

fedorkotov, 2021-03-12:
+1 from me. This will  be really nice to have. But keep in mind though that when used to select innermost or outermost of some closed curves this selector will work in many cases but not always. See counterexample [here](https://github.com/CadQuery/cadquery/issues/684#issuecomment-797482819)

marcus7070, 2021-03-17:
Done in #688.

closed 2021-03-17 by marcus7070