CadQuery/cadquery #650: debugging OCCT errors

mbway, 2021-02-20:
very often I'm finding that operations that work for some shapes fail when called on other shapes, so to get the result I want I'm trying to create the same shape or similar shapes in different ways until the command works. This is extremely frustrating and time consuming since the only feedback is `OCP.StdFail.StdFail_NotDone: BRep_API: command not done`.

Is there any way to obtain more information about what caused the failure, and perhaps since cadquery also controls the native bindings, perhaps there could be a way to trace what calls are made to OCCT and with which data so that problems can be examined in isolation (although if the API is stateful like cadquery then perhaps this wouldn't be useful)?

adam-urbanczyk, 2021-02-21:
`cq.Shape.exportBrep` gets you the internal (AFAIK lossless)  format of OCCT. You can use it in `draw.exe` to e.g. submit bugs on the OCCT tracker. We might want to implement `cq.Shape.importBrep`, but for now you can use OCP calls to get there (`OCP.BRepTools.BRepTools.Read_s`).

NB: CQ-editor only displays part of the stack trace, you can run the code in the embedded console or just in (i)python to get the full version. This will tell you which high-level OCCT operation is failing, but that is not always very helpful.

If you find a good way to debug OCCT issues do let us now.

fedorkotov, 2021-03-05:
@adam-urbanczyk Exporting BREP and stack trace are good but optional logging of all OCCT API calls made by CadQuery would be a even better. Ideally C++ program demonstrating an issue should be easy to reconstruct from such a log. This would help in reporting bugs with relevant examples to OCCT developers. I suspect this would also help you and other CadQuery developers in debugging efforts.

dcowden, 2021-03-05:
I agree with the idea of a trace. It would be super-helpful because we could ask for that as a standard bug report requirement. That's a great idea @fedorkotov

adam-urbanczyk, 2021-03-05:
I'm not sure that I understand how would a log of OCCT calls help in debugging when you have a complete python code you can debug. Or are you proposing tracing through the C++ code?

You can easily construct a MWE by dumping the breps at the offending OCCT call and then recreating them via draw. AFAIK this is the preferred format for a OCCT bug reports.

dcowden, 2021-03-05:
@adam-urbanczyk whats an MWE?

I've worked in several projects where 'layers' are involved-- in this case, python->OCCT, in other cases various layers of software. One example would be a enterprise system written in java that then runs a bunch of SQL.  A 'trace' includes debug data from the original code, and the SQL that is run, its results, and the results from the java layers, layed out on a strict timeline ( so you know what order various stuff happened in ).

Sure, you can log out just the SQL, and you can solve a lot of problems with that. but having the time-ordered results from each layer is superior, because you can observe communication problems between the layers, timing nuances, etc.

To me, this seems similar. If you could see the python code, and what OCCT methods it calls, in the order it called them, that's more powerful than dumping a brep, because i can see the precise order in which the calls were made.

adam-urbanczyk, 2021-03-05:
MWE = Minimal Working Example.  Again,I don't get what is the difference w.r.t. running in the python debugger. If someone wants to contribute  debugging code, sure let's take a look. If we need to report a bug on the OCCT tracker a MWE is what they are usually interested in.