CadQuery/cadquery #651: segfault when sweeping a circle along a triangle

fischman, 2021-02-21:
`cq.Workplane("XY").circle(0.5).sweep(cq.Workplane("XZ").polygon(4, 4))` causes CQ-editor 2.1 to segfault.

adam-urbanczyk, 2021-02-21:
In principle the CAD kernel should never segfault, but you are trying to perform some very weird operation. 

[image]

closed 2021-02-21 by adam-urbanczyk

fischman, 2021-02-23:
@adam-urbanczyk I hear you about prioritization, and that's fine, but FWIW your rendering suggests I'm trying to sweep a circle along a square in the same plane, which is not what my example is doing (sweeping a circle along a square in an orthogonal plane).

In case it's interesting, moving the circle avoids the crash and renders half the intended sweep, but not the other, presumably b/c the normal fails to flip correctly at a sharp corner:
`cq.Workplane("XY", origin=(2,0,0)).circle(0.5).sweep(cq.Workplane("XZ").polygon(4, 4))`

[image]

adam-urbanczyk, 2021-02-23:
If it helps, this works (though produces something slightly different):
```
path = cq.Workplane("XZ").transformed((0,0,45)).polygon(4, 4)
profile = cq.Workplane("XY", origin=(3,0,0)).circle(1)

res1 = profile.sweep(path,transition='round').translate((10,0))
res2 = profile.toPending().sweep(path,transition='right').translate((20,0))
```

fischman, 2021-02-23:
That's helpful, thanks!