CadQuery/cadquery #653: Show_object() does not work

hexad, 2021-02-23:
I am just getting started with cadquery. I am using cadquery as a FreeCAD plugin.
The first thing is i wanted to import a STEP file.

This is my code:
`import cadquery as cq
myPart = (cq.importers.importStep('/home/user1/Desktop/STEP_files/Cushion_Part_Test_v2_STP.step')
show_object(myPart, options={"rgba":(204, 204, 204, 0.0)})
`
However this does not work. 
I am getting:
`('invalid syntax', ('<cqscript>', 10, 1, 'show_object(myPart, options={"rgba":(204, 204, 204, 0.0)})\n'))`

However, i searched massively on the internet and this is the syntax that everyone uses.
Also experimented with passing no options, or using show(). But nothing worked.

Maybe this is a bug?

jmwright, 2021-02-23:
The FreeCAD workbench only supports CadQuery 1.x, so I would recommend moving to [CQ-editor](https://github.com/CadQuery/CQ-editor/releases/tag/0.2) unless you have to use the FreeCAD workbench for some reason.

hexad, 2021-02-23:
Thanks. This is the first time i see CQ-editor.
The only ways i knew to run cadquery were 1) from inside FreeCAD 2) with coda install 3) as a docker container with jupyter.
What is the version that works with CQ-editor?

hexad, 2021-02-23:
Never mind, i saw that you can do it with coda. I searched docker hub and i saw no images with it (used search term 'cq-editor')?
But perhaps this is a smaller package than cadquery, maybe it's not bering reffered. Is it available on some docker image online?

adam-urbanczyk, 2021-02-23:
You can install it via conda from the cadquery channel or use this prepackaged version (pyinstaller based): https://github.com/CadQuery/CQ-editor/releases/tag/0.2

hexad, 2021-02-24:
Thank you.
I installed CQ-editor from the prepackaged version.
However, the error persists here:
`import cadquery as cq
myPart = (cq.importers.importStep('/home/user1/Desktop/STEP_files/Cushion_Part_Test_v2_STP.step')
show_object(myPart, options={"rgba":(204, 204, 204, 0.0)})`

gives me:
`Syntax Error: invalid syntax line 3`

Also tried: `show_object(myPart)`

I don't understand why this isn't working? Perhaps loading STEP files is flawed on cadquery?

jmwright, 2021-02-24:
@hexad You haven't closed the parentheses around the `cq.importers.importStep` call.

hexad, 2021-02-24:
@jmwright Thanks! That was a careless omission. Issue solved. Maybe the ticked should be deleted since it is not related directly to cadquery?

jmwright, 2021-02-24:
@hexad I think it's fine to just close it and not delete it.

closed 2021-02-24 by jmwright

hexad, 2021-02-24:
Thank you for your support!